Optiboot/optiboot #127: Migrate Arduino Hardware Core to 1.6 format

swiftgeek, 2015-05-01:
https://github.com/arduino/Arduino/wiki/Arduino-Hardware-Cores-migration-guide-from-1.0-to-1.6

WestfW, 2015-05-09:
I think the only part relevant to optiboot is the location of the boards.txt file, and I think that that's been incorrect (or at least non-optimal) for quite a long time.

swiftgeek, 2015-05-09:
boards.txt + tree structure + i guess many missing files that could be most probably copied over from arduino/avr

WestfW, 2015-08-07:
Various commits:
https://github.com/Optiboot/optiboot/commit/988c239889e90feb100acb3682523804ae22f488
https://github.com/Optiboot/optiboot/commit/bfb3d5fc33a495e33a6cc591e2c4b7820aeb8710
https://github.com/Optiboot/optiboot/commit/53536624f2ed98c6a059ab6fef9e339ab764e8d0
https://github.com/Optiboot/optiboot/commit/8169eb5fc69d03cfef3da83e526dcfe059ade47b

WestfW, 2018-07-29:
I think I'm going to decline to modify the directory structure of the repository just because it no longer maps onto the directory structure used by the Arduino IDE itself.
Instead, the release.sh script will create appropriate directories and/or installation-controlling "whatever" files (currently json...)

closed 2018-07-29 by WestfW