Optiboot/optiboot #132: Add ATmega32u4 support

WestfW, 2015-05-21:
this will require fixing (or working around) the use of USTATbit0 for "all" uarts, since the 32u4 has only "uart1"

dmjlambert, 2015-07-17:
You are the one who reminded me of the importance of keeping things simple.  Thank you.  A few weeks ago after I sent you a pull request for that convoluted idea of mine, I went back to the drawing board.  As it turns out, the easy solution just requires a Makefile.custom entry.  

http://m.instructables.com/id/How-to-upload-sketches-to-Pro-MicroLeonardo-via-se/step4/Building-Optiboot-for-ATmega32U4/

dmjlambert, 2015-08-22:
I need to retract my previous comment.   I previously had just added to the Makefile.custom
```
atmega32u4: TARGET = atmega32u4
atmega32u4: RXEN1  = RXEN0
atmega32u4: TXEN1  = TXEN0
atmega32u4: UCSZ10 = UCSZ00
atmega32u4: UCSZ11 = UCSZ01
atmega32u4: UDRE1  = UDRE0
atmega32u4: RXC1   = RXC0
atmega32u4: FE1    = FE0 
atmega32u4: U2X1   = U2X0
atmega32u4: MCU_TARGET = atmega32u4
atmega32u4: CFLAGS += $(COMMON_OPTIONS) -DUART=1
atmega32u4: AVR_FREQ ?= 16000000L
atmega32u4: LDSECTIONS  = -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe
atmega32u4: $(PROGRAM)_atmega32u4.hex
atmega32u4: $(PROGRAM)_atmega32u4.lst
```

And then I compiled with a command like this on Mac
```
make OS=macosx ENV=arduino LED_DATA_FLASH=1 LED_START_FLASHES=3 LED=B0 BAUD_RATE=57600 atmega32u4
```

But I can no longer get that to work at all.   I have no idea why, because it definitely worked for me before.   So please disregard my comment from Jul 17, as I need to change the Instructable.

tzarc, 2016-06-11:
It can indeed be done with Makefile.custom - here's what I threw into the file for the base variant and an 8MHz version I've been using:

    atmega32u4: TARGET = atmega32u4
    atmega32u4: MCU_TARGET = atmega32u4
    atmega32u4: CFLAGS += $(COMMON_OPTIONS) -DUART=1 -DU2X0=U2X1 -DRXEN0=RXEN1 -DTXEN0=TXEN1 -DUCSZ00=UCSZ10 -DUCSZ01=UCSZ11 -DUCSZ02=UCSZ12 -DUDRE0=UDRE1 -DRXC0=RXC1 -DFE0=FE1
    atmega32u4: AVR_FREQ ?= 16000000L
    atmega32u4: LDSECTIONS = -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe
    atmega32u4: $(PROGRAM)_atmega32u4.hex
    atmega32u4: $(PROGRAM)_atmega32u4.lst

    atmega32u4_8MHz: TARGET = $@
    atmega32u4_8MHz: CHIP = atmega32u4
    atmega32u4_8MHz:
    	$(MAKE) $(CHIP) AVR_FREQ=8000000L LED=B5 BAUD_RATE=38400
    	mv $(PROGRAM)_$(CHIP).hex $(PROGRAM)_$(TARGET).hex
       	mv $(PROGRAM)_$(CHIP).lst $(PROGRAM)_$(TARGET).ls

It's probably not the nicest way of getting it working, but it's effective.

lbovay, 2017-09-01:
I'm in need of some extra space on my 32u4 and found this. Does anyone have an up to date version for the 32u4, possibly @ 16 MHz ?

dmjlambert, 2017-09-02:
tzarc mentions above what to do to compile one using Makefile.custom.   A compiled and ready-to-use hex file is on the Instructables article I mention in the comments above.

pawisoon, 2019-02-03:
Is there anybody who successfully compiled a bootloader for 8MHz version of Atmega32u4? 
If there is anybody with working bootloader please share it with us :) 
I tried to do it myself but I'm receiving this error message:
```
➜  optiboot32u4 make OS=macosx ENV=arduino LED_DATA_FLASH=1 LED_START_FLASHES=3 LED=B0 BAUD_RATE=57600 atmega32u4     
avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 57600, Real: 57142, UBRRL = 34, Difference=0.7%
../../../../tools/avr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega32u4 -DF_CPU=16000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3   -DLED_DATA_FLASH=1 -DLED=B0      -DUART=1 -DU2X0=U2X1 -DRXEN0=RXEN1 -DTXEN0=TXEN1 -DUCSZ00=UCSZ10 -DUCSZ01=UCSZ11 -DUCSZ02=UCSZ12 -DUDRE0=UDRE1 -DRXC0=RXC1 -DFE0=FE1   -c -o optiboot.o optiboot.c
../../../../tools/avr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega32u4 -DF_CPU=16000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3   -DLED_DATA_FLASH=1 -DLED=B0      -DUART=1 -DU2X0=U2X1 -DRXEN0=RXEN1 -DTXEN0=TXEN1 -DUCSZ00=UCSZ10 -DUCSZ01=UCSZ11 -DUCSZ02=UCSZ12 -DUDRE0=UDRE1 -DRXC0=RXC1 -DFE0=FE1 -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot_atmega32u4.elf optiboot.o 
/Users/pawisoon/Downloads/Arduino 2 copy.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld: section .version loaded at [00007ffe,00007fff] overlaps section .text loaded at [00007e00,00008009]
collect2: error: ld returned 1 exit status
make: *** [optiboot_atmega32u4.elf] Error 1
rm optiboot.o
```

WestfW, 2019-02-06:
The "overlaps section" error means that the bootloader binary is too big to fit in 512 bytes.
You can try turning off some of the newer features (or LED_DATA_FLASH) to see if that is sufficient, or turn on BIGBOOT and adjust the section start addresses.)
(I didn't look carefully enough to see WHY 32u4 is so much bigger, but it is...)

sv1, 2020-05-27:
> Is there anybody who successfully compiled a bootloader for an 8MHz version of Atmega32u4?
> If there is anybody with working bootloader please share it with us :)
> I tried to do it myself but I'm receiving this error message:
> 
> ```
> ➜  optiboot32u4 make OS=macosx ENV=arduino LED_DATA_FLASH=1 LED_START_FLASHES=3 LED=B0 BAUD_RATE=57600 atmega32u4     
> avr-gcc (GCC) 5.4.0
> Copyright (C) 2015 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> 
> BAUD RATE CHECK: Desired: 57600, Real: 57142, UBRRL = 34, Difference=0.7%
> ../../../../tools/avr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega32u4 -DF_CPU=16000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3   -DLED_DATA_FLASH=1 -DLED=B0      -DUART=1 -DU2X0=U2X1 -DRXEN0=RXEN1 -DTXEN0=TXEN1 -DUCSZ00=UCSZ10 -DUCSZ01=UCSZ11 -DUCSZ02=UCSZ12 -DUDRE0=UDRE1 -DRXC0=RXC1 -DFE0=FE1   -c -o optiboot.o optiboot.c
> ../../../../tools/avr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega32u4 -DF_CPU=16000000L  -DBAUD_RATE=57600 -DLED_START_FLASHES=3   -DLED_DATA_FLASH=1 -DLED=B0      -DUART=1 -DU2X0=U2X1 -DRXEN0=RXEN1 -DTXEN0=TXEN1 -DUCSZ00=UCSZ10 -DUCSZ01=UCSZ11 -DUCSZ02=UCSZ12 -DUDRE0=UDRE1 -DRXC0=RXC1 -DFE0=FE1 -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot_atmega32u4.elf optiboot.o 
> /Users/pawisoon/Downloads/Arduino 2 copy.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld: section .version loaded at [00007ffe,00007fff] overlaps section .text loaded at [00007e00,00008009]
> collect2: error: ld returned 1 exit status
> make: *** [optiboot_atmega32u4.elf] Error 1
> rm optiboot.o
> ```

Same problem here, even if I set LED_DATA_FLASH=0

sv1, 2020-05-27:
Enabling BIGBOOT and adjusting section start address to 0x7ce0 compiles successfully.

Virtual-Java, 2021-09-29:
This issue has been solved by my variant of Optiboot supporting all 8-bit AVR-USB-MCUs known to me yet.
I was able to successfully compile for all supported target controllers without exceeding 1 kB.

Since I started a pull request, the new feathers should be supported by the official Optiboot release very soon.

Everybody interested in further extensions of Optiboot can have a look at https://github.com/Virtual-Java/optiboot.

Virtual-Java, 2021-09-29:
Maybe an interesting comment:
I realized that the hexfile created for ATmega32u6 is much smaller in size than the one for ATmega32u4.

Sometimes I suppose there is an bug in the compiler which creates some redundant push/pop and mov instructions.
Consequently the led has to be disabled for the Atmega32u4 to fit into the 512 Bytes boot section (results in 480 Bytes with gcc5.4). Otherwise BIGBOOT has to be enabled.
When I've done my analyses and hopefully found a reason for this issue I will post it here.

Do you think it is possible to save some memory by improving the led functions?
However they consume 44 Bytes of flash for the ATmegaXYu2 USB-MCUs.
(438 Bytes when LED_START_FLASHES = 0 compared to 482 Bytes when LED_START_FLASHES = 3)

WestfW, 2021-10-01:
> I realized that the hexfile created for ATmega32u6 is much smaller in size than the one for ATmega32u4.

I did not observe this difference when I compiled your branch here?

```
 ll optiboot_atmega32u*_UART1_115200_16000000L_BB.hex 
-rw-rw-r--  1 staff  1401 Sep 30 18:51 optiboot_atmega32u2_UART1_115200_16000000L_BB.hex
-rw-rw-r--  1 staff  1397 Sep 30 18:51 optiboot_atmega32u4_UART1_115200_16000000L_BB.hex
-rw-rw-r--  1 staff  1401 Sep 30 18:51 optiboot_atmega32u6_UART1_115200_16000000L_BB.hex
```

> Do you think it is possible to save some memory by improving the led functions?
There is "LED_START_ON" which is sort-of supposed to address this.
I did some experiments with using delay loops instead of the timer, but the results were not that promising (It saved 6 bytes.  It takes a 24bit counter to produced a reasonable delay loop.)

WestfW, 2021-10-04:
Virtual-Java's Pull Request has been merged.

closed 2021-10-04 by WestfW