Optiboot/optiboot #136: Protection from Bootloader Executing during Application

vicatcu, 2015-05-29:
This is not a hypothetical issue, so maybe there's a better medium for exploring it than a GitHub issue, but I recently posted this question to the EE stack exchange: http://electronics.stackexchange.com/questions/172682/avr-bootloader-protection. 

One of the answers struck me as interesting, and possibly helped me shed some light on the age old question of why does Optiboot clear the MCUSR before launching the application, yet still left me a bt puzzled. The answer led me to believe that one reason to clear the MCUSR at the exit from the bootloader is so that the bootloader can subsequently detect when it's being executed when it shouldn't be. Obviously that's a bad thing as the bootloader is the only code running that can do semi-permanent damage to the application code, since it has the privilege of executing SPM instructions.

It struck me, however, that as far as I can tell, the current source for Optiboot doesn't take advantage of this insight, so (maybe?) it's not safe from errant application code invoking the bootloader. Am I missing something, like do the bootloader support functions in avr-libc exploit this tidbit natively to only issue SPM instructions if the MCUSR is not cleared? Would it even be possible for us to enforce such a check ourselves, or would it entail a rewrite of the avr-libc bootloader support functions to incorporate this convention?

WestfW, 2015-05-30:
The main reason to clear MCUSR seems to be that the bits are sticky.  If you don't clear it, the register will show the logic OR of all the reset causes since the last poweron, which is probably not very useful.
Also, since optiboot uses the WDT to reset state before entering the application, it would end up saying that the reset cause was WDRF even if the user had never enabled WDT, which wouldn't be good either.

Optiboot certainly doesn't try to detect wayward jumps into its code; I'm not even sure that's possible (give the number of cycles available in the timed sequence.)  In fact, there's an open issue in the other direction - to explicitly allow calls to the bootloader code so that an application can write program memory.

Did you see http://hackaday.com/2014/07/05/overwriting-a-protected-avr-bootloader/ ?

vicatcu, 2015-05-31:
@WestfW, right I got it, there are perfectly legitimate reasons to intentionally jump to the bootloader from the application. I've just had a couple of unexplained experiences recently where application code seems to have gotten corrupted and re-burning over USB / Serial restores the chip to working order. I can't think of a way this could happen other than inadvertent calls to the bootloader. 

That being said, I appreciate that there are competing (and pretty neat) interests at play, and Optiboot can't be all things to all people. Also I recognize that there's no substitute for writing application code that doesn't do the unexpected, but if there's ways to insulate from that possibility (we're talking about C, here, after all) I'm certainly interested. I suppose I'm in favor reliability over flexibility at the moment, but maybe I'm too paranoid and something else is to blame for my mysteries :-).

MarkG55, 2015-06-03:
If you're really concerned about this possibility, I think that the only way to be sure would be to just use an In System Programmer to upload your code and not have the bootloader at all.  Also make sure that the appropriate lock bits are set as well.  

At least if you try this and you find that your application code is still being corrupted, you'll know that it's nothing to do with the bootloader...

closed 2015-08-07 by WestfW