Optiboot/optiboot #139: Adding documentation

majekw, 2015-05-31:
Hello.

Is it a 'proper' way for adding documentation?
In repository there is a readme.txt, but according to it main source of documentation is wiki.
Is it a way to submit patches or new files to wiki?

I don't know how it's organized and I want to contribute what I learned so far about Optiboot (for example separate chapter about Virtual boot partition).

WestfW, 2015-05-31:
The wiki is "sort-of" a separate repository, and can be cloned (and probably get pull requests) via https://github.com/Optiboot/optiboot.wiki.git
There's an option to make the wiki section fully public so that anyone can edit it, but that seems excessively permissive to me.

NicoHood, 2015-06-01:
I'd suggest to add a note to the readme that the wiki exists.
https://github.com/Optiboot/optiboot/wiki

I wanted to recompile optiboot but I am a bit lost how to change things like baud rate, mcu etc.
Some information about this would be nice. And where to get informatio about the virtual boot partition etc.

BTW: make all doesnt work for me with avr-gcc 4.8.2 (havent tried any older)

WestfW, 2015-06-02:
readme.md modified.

>> make all doesnt work for me with avr-gcc 4.8.2
There is no "make all"; there is a "makeall" shell script.  If that's not working, open a new issue.
(I'm still back at 4.8.1)

NicoHood, 2015-06-02:
ah! makeall seems to work.
4.8.2 is standard in ubuntu but it has some bugs I read some month ago.

My Arduino nano works with baud 250000 at the serial monitor of the Arduino IDE.
The bootloader however does not work this speed. 500k same result.
I compiled the bootloader on my own and also tried .hex files from some blogs.
Of course I changed the boards.txt to the new baud rate, it was correctly used in the verbose output of avrdude in the IDE.
But my Arduino nano didnt want to be programmed with 250k or 500k, even though those blogs say it should work and the Serial monitor works for 250k (possible with the new nightly IDE).

Do you think my fake FTDI is bad quality or why does this fail?

majekw, 2015-06-21:
Bill, how about moving/adding all documentation into separate folder, for example docs/ ?
There should be no issues with wiki access, recent documentation will be always a part of source/package and this could be also updated by normal pull requests.

majekw, 2015-06-24:
I tried to push some changes to cloned wiki, but it failed (permission denied). So, this way doesn't work.

I've written whole new page about Virtual Boot Partition, so @WestfW , I hope you'll find a way to put it into wiki.
It's available here: https://github.com/majekw/optiboot/wiki/Virtual-Boot-Partition

WestfW, 2015-06-25:
Added manually.  Thanks!   Odd that it doesn't seem to have the same request model as the code side.
BillW

> I tried to push some changes to cloned wiki, but it failed (permission denied). So, this way doesn't work.>> I've written whole new page about Virtual Boot Partition, so @WestfW , I hope you'll find [...]

majekw, 2015-07-08:
@WestfW, another small documentation update requested in #106: 
https://github.com/majekw/optiboot/wiki/CompilingOptiboot/_compare/f34d8ea3271ad16fd16fbb200dfc908ff5836c2a...76108b5b0d40a4e777cf4bac31665b8bdc3ad9b6

WestfW, 2018-08-07:
I think I shall experiment with moving the wiki pages into the main repository, which I think can do markdown and follow links and whatnot, though it might require changing WikiStuleLinks to real web-links...
That way regular pull requests will work.

majekw, 2018-10-07:
Hello, @WestfW 
I see you put wiki pages into main repository which is great news. Do you plan to merge docs/ and Wiki/ directories? As wiki is no longer a wiki, to I think that moving everything to docs/ dir would be natural direction :-)