Optiboot/optiboot #140: Fast start/ MCUSR issues and regression

WestfW, 2015-06-01:
It looks as though the changes to the no-wait modification need to be re-examined.
They probably break the "hold reset while powering on" fix for certain brick situations, because that results in MCUSR having PORF+EXTRF set. That used to run the bootloader, but now will skip it. :-(

Edit made in https://github.com/Optiboot/optiboot/commit/c51c79299739a1d81f2d87129707bb71fbc616ca

majekw, 2015-06-01:
I've made a table which could help think about right solution. I skipped BODF flag as it doesn't make much difference.

PORF | EXTRF | WDTF | prio | flags to clear | should run | old run | new/current run | comments
---------|-----------|-----------|-----------------|-------------------|------|-----------|------------|----------
0 | 0 | 0 | | | app | app | app |
0 | 0 | 1 | | | app | app | app |
0 | 1 | 0 | 3 | | boot | boot | boot |
0 | 1 | 1 | 2 | WDTF | app/boot? | boot | app | first start of application by wdt from optiboot
1 | 0 | 0 | | | app | app | app |
1 | 0 | 1 | | | app | app | app |
1 | 1 | 0 | 1 | | boot | boot | app | emergency sequence
1 | 1 | 1 | 1 | | boot | boot | app | emergency sequence

I could update this table as comments arrive.

majekw, 2015-06-01:
I think that 2 questions need to be answered:
1. What should be run in PORF=0, EXRF=1, WDRF=1 case?
In most cases: application with WDRF cleared (WDRF isn't triggered by application, but by bootloader itself).
But in less likely case (1), safer could be to run bootloader.

2. Should we clear some flags after running bootloader code?
It's speed vs. accuracy question. If we clear for example PORF and EXTRF after emergency sequence, we could speed up booting after second reset or even from application watchdog if application doesn't clear PORF. In controversial case from question 1., it could also help.
Of course we could save MCUSR in some register for application, but application usually don't expect flags in such strange places and must be modified.

MarkG55, 2015-06-01:
If you take a look at my suggested change, it doesn't suffer from this problem - if EXTRF flag is set during power on, my code will run the bootloader.  This does mean that you won't get a fast boot if your hardware causes EXTRF to be set during power up, but there's no way to get a fast boot in that situation without losing the ability to run the bootloader by holding external reset during power on.

One other possible minor issue with my code is that if EXTRF is set and the user code doesn't clear it, a later brown out reset won't get a fast boot.  I don't see this as a major problem myself - it's better to err on the side of running the bootloader and incur a small time penalty than to possibly have a situation where the device is bricked because the bootloader doesn't run when it should.

I agree with the logic of only clearing WDRF if the watchdog was due to the bootloader - I'll post a modified version of my code over in the other issue:
https://github.com/Optiboot/optiboot/issues/97

majekw, 2015-06-02:
According to truth table, perfect boot sequence should look something like this:
`````
c = MCUSR;
if ( !(c & _BV(EXTRF)) ) appStart();  // EXTRF=0
if ( !(c & _BV(PORF) ) {  // EXTRF=1, PORF=0
  if (c & _BV(WDRF) ) {
    MCUSR&=~(_BV(WDRF)); // clear WDRF for EXTRF=1, PORF=0, WDRF=1
    appStart();
  }
}
...bootloader code follows
`````
It could be easily optimized, but I leave it that way for readability. Implementation is another story :-)
To speed up booting we should think about clearing EXTRF flag in some situations, but not in all as it would break detecting watchdog from bootloader.

MarkG55, 2015-06-03:
I can see a problem with that code:  If you're not clearing MCUSR, and the user code is also not clearing MCUSR, after an external reset, you'll be stuck in a constant loop running the bootloader with the only way to get back into user code being to power cycle the board.

Is there any reason why you're not willing to consider going with my suggested code changes?  I'll post another modified version over in the other issue:
https://github.com/Optiboot/optiboot/issues/97

WestfW, 2015-06-03:
>> Is there any reason why you're not willing to consider going with my suggested code changes? 

I am considering it.  That's why we're talking!
But I've got four different suggested patches for a "problem" that affects a small minority of users, and at least two past instances where the code was determined to be wrong.  Plus, "every byte counts."  So I'm trying to be really careful about not breaking a "common" usage scenario to acomadate people who could patch their own bootloader anyway.  And I don't have a test fixture that will generate all of the cases that I want to test. :-(

MarkG55, 2015-06-03:
Hi Bill,

Sorry - I wasn't meaning that so much at you as to majekw who is suggesting numerous different versions that obviously haven't been tested.

You could look at it from the point of view that my suggested code changes fix this (potentially serious) regression (and pretty much eliminates the need for "reset during power on") with the bonus of keeping MCUSR intact for the minority of users that do need/want it.  It also adds in the capability to jump into the bootloader as requested in another issue:
https://github.com/Optiboot/optiboot/issues/52

I appreciate that every byte counts, so maybe it makes sense to use a conditional define on whether to allow jumping into the bootloader, but I don't think that my other changes really add much to the size of the bootloader.

I have tested the code used in my fork, but I can understand you wanting to test it for yourself of course.

majekw, 2015-06-03:
MarkG55: please, it could be easier to talk about problem than 'my code' or 'your code'. I don't care who's code will solve problem as long as it really do this in clean way. Bill was also very patient - he could just revert commit to solve this regression, but he waited for us for better solution (I hope :-) ).

From what I can see, your current proposition is almost the same as in the table. Only one difference is that in case of PORF=1 EXRF=1 and WDRF=1 application is started instead of bootloader.
It still suffers from (1), but should be recoverable by 'emergency sequence' assuming that WDRF should be reset during power on.

Bill, I would give a try to MarkG55's solution. If it's size overhead will be too large, I could try to rewrite it in assembler.
Test scenario: application which enable watchdog with very short timeout and do nothing more.

WestfW, 2015-06-06:
So here's my plan.
I want to get Optiboot to a nice safe point where I can tag/release "v7.0" (or something), as a IDE 1.6.4 compatible .json file.
So I plan to fix the regression and apply some variation of the Streef patch that puts more accurate values in r2, but leaves MCUSR itself clear.  This is very low-risk, and compatible with the way that 4.4 behaves.
I'll keep the "don't clear MCUSR" issues and discussion open for possible inclusion after that.

MarkG55, 2015-06-06:
Hi Bill,

I understand the desire to go with a safe option...

I would be interested in seeing the variation of the Streef patch that you're thinking of going with since by my reading of the code, the current version of that patch with leave r2 undefined after a true watchdog reset.

Personally, I don't think that there are that many combinations that it is difficult to logically go through them all and see which solution works the best.

majekw: Sorry I didn't see your reply - for some reason, I didn't get an email from GitHub (just checked, and yep - was in my spam folder - my fault for not logging in to see if there was activity rather than assuming that things had gone quiet)

I agree that it shouldn't be about "yours" vs "my" code - my apologies if I've been too antagonistic.  We should go with whichever code works the best.

In the case of PORF=1 EXRF=1 and WDRF=1, yes my code would start the application instead of the boot loader.  I don't think that would be a problem though - the only likely scenario that I can think of where that could occur would be if the user code didn't clear MCUSR and a watchdog reset occurred - in which case it would make sense to jump straight to the application code.

If an application were to enable the watchdog with a short timeout and then immediately crash, my expectation would be that the following would occur with my code:

1.  Power On.
2.  Bootloader jumps to application after clearing WDRF (if set).
3.  Application Enables watchdog.
4.  Application crashes.
5.  Watchdog reset - jump to step 2.

The vast majority of the time (even with a short watchdog timeout) will be spent at step 4 - waiting for the watchdog to reset the controller.

To recover from the above - even after looping for any number of times, it would be just a case of uploading the firmware as normal.  There would be a (very) small window of time during which the upload would fail - the time between the watchdog reset occurring (with WDRF being set) and the bootloader clearing WDRF and starting the application.  Once WDRF is cleared, an external reset will correctly cause the bootloader to be run.  If there was enough motivation, I'm sure that we could work out the amount of time between WDRF being set and then being cleared so that we could calculate the percentage failure rate, but I'd expect that it would be a very low chance - and shouldn't be substantially different to any other code that does clear MCUSR anyway.

Again, I think that the following code is quite safe:
```
MCUSR = ~(_BV(WDRF));
```

In my experience, it's quite safe to do and I think that it would be a very poor design choice at Atmel if it was possible to set the bits in MCUSR in application code - since the bits are sticky through a reset, bad code could set some or all bits before a watchdog reset occurs and MCUSR would then contain undefined flags.

With that being said, the data sheet doesn't explicitly say that it's impossible to set the flags in code, but by my reading of what is written, it's very much implied.  Interestingly, the sample code in the data sheet does follow the more safe option - both in C and assembler.

In the end, I think that when you're doing the utmost to reduce code size, it's a legitimate optimization, but I can understand the reluctance to go with something that could potentially be unsafe.

Bill:  I understand that since you're the maintainer, it's your call on which code to go with.  I promise not to get into (too much of) a huff if you decide not to go with my suggestions - either straight away or even further down the track.  Not that it would (or should) make a difference anyway of course! ;-)

majekw, 2015-06-09:
MarkG55:

1. 'Loop' case could be recovered by powering up board with reset pressed (WDRF should be cleared), so it should be fine - no permanent chip bricking :-)
2. Yes,`MCUSR = ~(_BV(WDRF));` gives one instruction (2B) less of code. Adding comment in code to this statement, that it was done deliberately because of sth would be nice and time saving for future code readers :-)
3. I think that now it could be much better if you could catch up with master and provide clean patch - I think that Bill could be also less reluctant then to integrate this feature/fix :-)

Bill:
I would also be glad to see what Steef-like patch you have in mind.

MarkG55, 2015-06-10:
OK, I'll work on creating a clean patch for Bill to hopefully consider.

I'm actually away at the moment - I don't have much time to work on it and I don't have any hardware with me to test it on, so it'll probably be a few days at least before I get to this.

majekw, 2015-06-27:
I've made patch using @MarkG55 code here: https://github.com/majekw/optiboot/commit/69e7b740f0757edd6b5bd9b8a76c9f73b816ff45
I briefly tested it on Atmega328 and it works.
I didn't make pull request to get chance for @MarkG55 to push himself it's own code :-)

wintersandroid, 2015-06-27:
I am having a problem with the fast-start, I have a custom board based on the atmega328p, and trying to use the optiboot because I need the extra 1.5k it seems.

In my program it has a serial command processor and while running it looks for the bootloader signature characters ( zero then space) before jumping to the bootloader. In the past with other bootloaders this worked good..

With optiboot, it seems no-matter what I do (watchdog e.t.c) the bootloader jumps directly to the program without giving any chance to upload the new program.. Even poweroff and on doesnt give me time for the bootloader.

Can the fastboot be an option? at the moment I have to re-flash optiboot which erases the chip to be able to upload the program via serial, I use this method to be able to test the programs "hands-off" the hardware. (there is no reset button, the reset pin is tied to the ISP).

tried code
```
void SerialHandlerCheckBootload(uint8_t byte1, uint8_t byte2)
{
  if(byte1 == '0' && byte2 == ' ')
  {
    // boot loader required..
//    PCICR = 0;      // disable pin-change interrupts
//    PCMSK0 = 0;
//    PCMSK1 = 0;
//    PCMSK2 = 0;
//
//    TCCR1A = 0;
//    TCCR1B = 0;
//    OCR1A  = 0;  //  20MHz clock / 20000 = 1mS
//    TIMSK1 = 0;
//
//    goto *0x7E00; // optiboot.
   //goto *0x7800; // bootloader. standard 2k

// try watchdog for optiboot.
      cli(); //irq's off
      wdt_enable(WDTO_15MS); //wd on,15ms
      while(1); //loop
 
  }
}
```

MarkG55, 2015-06-28:
majekw: Sorry I haven't had a chance to get back to this - I've been snowed under with work the last couple of weeks and also had some illness in the family on top.

If you don't mind doing it, I'm happy for you to do a pull request with that change - thanks for taking the time to do it!

MarkG55, 2015-06-28:
wintersandroid:Because of the way that Optiboot works, I don't think that you'll ever be able to start it by using the watchdog.

You might want to try my branch of Optiboot:
https://github.com/MarkG55/optiboot

Amongst my changes in that branch, I added the capability to jump into the boot loader.  As long as you ensure that MCUSR is cleared, you can jump to the bootloader and it will run with an 8 second timeout before starting the application code.

majekw, 2015-06-28:
@wintersandroid : as MarkG55 said, using watchdog wont work to start Optiboot, regardless of fast start issue.

You need Optiboot with MarkG55 patch. It's available from his fork, or more recent code with all current fixes and new write-to-flash feature (and also with precompiled hex files) in my branch https://github.com/majekw/optiboot/tree/supermaster
I'll appreciate any feedback from my branch as this code is 'to be merged' soon.

So, when you have Optiboot with this patch, all you need to run bootloader is:
````
MCUSR=0;
goto *0x7E00;
````

MarkG55, 2015-06-28:
@majekw I've just checked your branch, and it doesn't look like you have included my changes to allow jumping into bootloader?

majekw, 2015-06-28:
@MarkG55 it's here:-) I just skipped setting longer watchdog timeout for this case because I treated it more like fix for 'fast start', not an additional feature. But it still runs bootloader for MCUSR==0.
I think that if 1s timeout will be issue, we could add more time for it later - now it's fix, and it seems that Bill is expecting fixes before features.

MarkG55, 2015-06-28:
@majekw ah - yes you're right!  Sorry!

wintersandroid, 2015-06-28:
@majekw

Have tried your code, came into an issue once I uploaded the bootloader, then powerd off (disconnecting the ISP) i didnt get a boot into program mode, LED was blank and I couldnt upload. I had to jumper the reset pin a few times to go into program mode.

Once I got the program on there, I was able to re-write it automatically, 

After the flash, the program is started and it gets into some sort of nasty crash loop and has to be powered off. I have not seen that before. I think there is some watchdog still running! I am looking into that now.

MarkG55, 2015-06-28:
@wintersandroid 

With my changes (and presumably also with the version now by @majekw) you will either need to jumper the reset pin to start the boot loader, or flash the firmware after uploading the boot loader - without powering the board off.

It might be that the crash loop that you're seeing is due to the stack pointer not being reset - try resetting it before jumping into the boot loader and see if that makes a difference?

wintersandroid, 2015-06-28:
@MarkG55 

Any idea on how to reset the stack pointer before Jumping?
I have tried 
SP=RAMEND before  goto *0x7E00; and I seem to get some sort of lockup - searching google for resetting the stack is not helping..

majekw, 2015-06-28:
@wintersandroid 

1. Because of 'fast start' using reset is now mandatory to enter bootloader (except jump from application), so what you describe in first paragraph is 'work as intended'.
One thing to think about is increasing initial timeout. It's 1s right now and it's a little too short for human.

2. Could you provide more details of this 'nasty crash'? Does your application works somehow?

3. Are you using watchdog in your application?

4. Are you using the same serial port to send commands to your application and to program chip? If yes, how are you dealing with sharing between 2 programs?

5. If you put SP=RAMEND before 'goto', it's definitely not a stack issue.

6. Do you have 'make' utility to compile Optiboot from sources?

7. Do you use Arduino (which version) for compilation and upload, or do you just use bare avr-gcc?

wintersandroid, 2015-06-28:
I think I will have to fork and make a version that doesn't have the fast start - (has it as an make option)..

I am using arduino-mk with the arduino libraries, for the application. I used make for making optiboot with my own configuration that matches the hardware - all done on the same port.

matthijskooijman, 2016-09-22:
I've also ran into this issue, needing a working watchdog timer *and* needing to know the reset reason to show an error message. I've dug into all the discussion about this topic and it seems there is no perfect solution here. I'll need a solution on the short term, so I'll end up implementing something regardless of what will be merged. Since I needed to sort out my thoughts about this, I wrote a bit of analysis on the subject, which I'll share here. Apologies for the length and limited structure, it's a bit of a braindump. Here goes.

It seems this issue is so complicated because:
 - We want to tell the application when a watchdog reset occurred
 - The watchdog reset is used to clean up after the bootloader ran.

For this combination to work, the bootloader must be able to distinguish between watchdog resets originating from the bootloader and from the application. This is regardless of *how* we tell the application about the reset reason (e.g. through `r2` or `MCUSR`).

Roughly, I think there are a few ways to achieve this:
 - #125 looks at the top part of the RAM, at what I believe should contain the return address of the first function called. This seems like a clean way, but apparently there are issues with different version of the compiler (not) allocating a stack frame. In general, this approach is probably tricky since it relies on both the compiler for the bootloader *and* the main program to behave in a certain way.
 - Generalizing the above approach, you could try to store a flag in memory or some register. However, the problem is again that you can't predict what the main program will do, and it seems impossible to set a flag in the bootloader that cannot be accidentally set in the main program as well. Using some (long) magic marker might make it sufficiently unlikely, perhaps.
 - #158 essentially uses the `EXTRF` flag in `MCUSR` to distinguish between that. Since, in that approach, only `WDRF` is ever cleared from `MCUSR`, and the bootloader only ever runs when `EXTRF` is set, the "cleanup after bootloader" case will certainly have `EXTRF | WDRF` set. The risk here is that, under some circumstances, the "watchdog reset from application" will also have these two flags set.
 - A variant on the previous approach (not yet discussed previously), would be to use the `EXTRF` flag to distinguish between the cases as above, but to also clear `MCUSR` before jumping into the application (as happens in the current code). This ensures that the application is always started with an empty `MCUSR`, and because the application can never enable bits in `MCUSR`, any watchdog resets coming from the application will not have `EXTRF` set, preventing the false positives.

I believe the approach in #158 is promising, though still has a few corner cases. However, as Bill indicated in [a previous comment](https://github.com/Optiboot/optiboot/issues/140#issuecomment-109534203), he wants to stick to the behaviour of clearing `MCUSR` for now, so the last option above seems promising to me. I'm not sure if it's perfect yet, but we'll find out when I further investigate it below.

I suggested clearing `MCUSR` before running the application, but should it also be cleared when proceeding into the bootloader?
 - An advantage of keeping `MCUSR` while the bootloader is running is that it provides a nice place to keep the reset flags and copy them into `r2` later. `MCUSR` will survive the cleanup-watchdog-reset when the bootloader is finished.
 - A possible complication of keeping `MCUSR` is that `WDRF` might end up being set twice (once for an application watchdog reset and one for the bootloader cleanup reset), but this is easily solved by never running the bootloader after an (application or bootloader) watchdog reset (which is what you'd want anyway, and this is also what #158 implements).
 - There might be additional corner cases when you do not clear `MCUSR`, related to multiple reset reasons occuring close together and/or reset occuring while the bootloader is running.
 - If you clear `MCUSR` (all flags except `EXTRF`, since that one is needed later to distinguish watchdog resets) while running the bootloader (as #125, though that clears it completely) you'd need some other way of storing the reset reasons throughout the running of the bootloader, that would survive the cleanup reset. I think #125 just sticks it into `r2` early and keeps it there, though I'm not entirely sure how it guarantees that the compiler won't overwrite `r2` again while the bootloader runs.

As for what should happen on a reset, that seems simple: When an external reset happens, the bootloader should always run. In all other cases, the application should run. However, "when a reset happens" does not necessarily equal "`EXTRF` is set". This seems an argument to clear `MCUSR` when the bootloader starts. If you do this, the `MCUSR` register should almost always reflect the most recent reset cause. Keeping `EXTRF` set while the bootloader runs shouldn't be a problem either, since when another external reset happens when the bootloader is running, the bootloader should run again anyway. When another reset (that would be brown out) happens while the bootloader runs, running the bootloader again should be ok as well. A watchdog reset while the bootloader is running should never occur (only when its done).

[... cut ...]
When a reset happens while the bootloader is executing, the behaviour is slightly different than before, since these resets will cumulate in the `MCUSR` register (where the previous approach would discard them, since `r2` was already assigned). Cumulating them actually seems sane, since then the application will get reset flags for each reset that occured since the application last executed.

The code of this version actually ended up pretty much identical to #158, except that this version keeps `MCUSR=0`. This would probably make it easy to migrate to that version later on.

I'll go for this last version, and perhaps provide a PR to further clutter the field of possible solutions :-)

MarkG55, 2016-09-22:
> #158 essentially uses the EXTRF flag in MCUSR to distinguish between that. Since, in that approach, only WDRF is ever cleared from MCUSR, and the bootloader only ever runs when EXTRF is set, the [...]

Can you explain under what circumstances you think that EXTRF and WDRF would be set together?

matthijskooijman, 2016-09-23:
> Can you explain under what circumstances you think that EXTRF and WDRF would be set together?

As IIRC you previously pointed out already: when the user application does not clear `MCUSR`, and then a watchdog reset occurs. But also, when a watchdog reset occurs very quickly followed by an external reset (before the application can clear `WDRF`).

MarkG55, 2016-09-23:
I personally don't see either of those as being an issue:

If an application is using the watchdog flags, then it really should be clearing MCUSR as soon as it reads it - otherwise it's going to be seeing multiple reset flags regardless of the bootloader in use.

Similarly, there should only be an extremely small period of time between a watchdog reset occurring and an external reset occurring for the second scenario to be an issue - again assuming that the application code is clearing MCUSR as soon as possible as it should.  A semi educated guess would be sub 1 millisecond but it might be somewhat more than that...

Maybe I'm missing something with this though, so please point it out if I am!

matthijskooijman, 2016-09-23:
> I personally don't see either of those as being an issue:

I was just identifying corner cases, not saying that they would be necessarily an issue. The tiny window of the second issue makes it pretty much a non-issue indeed. I agree that an application using the watchdog should take care of `MCUSR` as well, but given that there is no need with the current version of optiboot, I can imagine applications will not do it right now, so *not* clearing `MCUSR` can introduce compatibility issues, which I presume is (partly) why @WestfW proposed to delay that until a later version.

MarkG55, 2016-09-23:
OK, I think that I understand where you're coming from - for existing code that uses the current Optiboot method for passing the reset flags then, yes, this could be a breaking issue for that code.

I think that it should be kept in mind that any code that relies on the current Optiboot method would also be broken with a standard bootloader or even no bootloader.  My suggested changes make the reset flag handling much closer to the standard/no bootloader case and I don't see any way around that.  I still think that it's something worth doing - i.e. improving compatibility at the cost of potential short term problems - as long as the change is sufficiently well documented and advertised.

matthijskooijman, 2016-09-23:
Interesting comment about compatibility with running without a bootloader, I had not considered that. That is indeed an extra reason to not clear `MCUSR`. I'm not sure if it would be better to do that now, or leave it as a later breaking change, that's up to @WestfW.

In any case, I implemented and tested the version shown above, which seems to work well. The change is [available here](https://github.com/Devoteq/optiboot/commit/d6b6d36fd7c9c6248f6d280f537987088046d821), and I've also updated [my comment above](https://github.com/Optiboot/optiboot/issues/140#issuecomment-249021798) with the tested code.

Your comment about running without bootloader made me hesitant about just relying on `r2` being set by optiboot, so I've put the following snippet in my application, which should work with a MCUSR-clearing bootloader, a completely MCUSR-preserving bootloader and no bootloader (a bootloader that clears MCUSR to indicate WDR happened, as I think was suggested in a comment somewhere, would break this code, though).

```
// Startup value of MCUSR register. Put in .noinit, since it is filled
// by get_mcusr before gcc initializes global variables.
static uint8_t mcusr __attribute__((section(".noinit")));

// Optiboot clears MCUSR and puts the original value in r2. This bit of
// code runs as soon as possible during startup to preserve the value.
// However, if MCUSR is not cleared, just use that instead (future
// versions of optiboot might no longer clear MCUSR, or we might be
// running without a bootloader). Also, make sure to clear MCUSR and
// disable the watchdog, just in case.
// This uses inline assembly, since the compiler might generate code
// that uses the stack, or assumes r0 is set to 0, etc.
void get_mcusr(void) __attribute__((naked)) __attribute__((used)) __attribute__((section(".init0")));
void get_mcusr(void)
{
  asm(
    // Store MCUSR, or r2 if MCUSR is 0
    "in  %[mcusr_var], %[mcusr_reg]\n"
    "subi %[mcusr_var], 0\n"
    "brne mcusr_not_empty\n"
    "mov %[mcusr_var], r2\n"
    "mcusr_not_empty:\n"
    // Clear MCUSR
    "clr r1\n"
    "out %[mcusr_reg], r1\n"
    // Disable watchdog
    "ldi r16, %[change_enable]\n"
    "sts %[wdtcsr_reg], r16\n"
    "sts %[wdtcsr_reg], r1\n"
    : [mcusr_var] "=d" (mcusr)
    : [mcusr_reg] "I" (_SFR_IO_ADDR(MCUSR)),
      [wdtcsr_reg] "M" (_SFR_MEM_ADDR(WDTCSR)),
      [change_enable] "M" ((1 << WDCE) | (1 << WDE))
    : "r1", "r16"
  );
}
```

miketedeschi, 2016-11-29:
@majekw 
`You need Optiboot with MarkG55 patch. It's available from his fork, or more recent code with all current fixes and new write-to-flash feature (and also with precompiled hex files) in my branch https://github.com/majekw/optiboot/tree/supermaster
I'll appreciate any feedback from my branch as this code is going 'to be merged' soon.
So, when you have Optiboot with this patch, all you need to run bootloader is:
cli();
MCUSR=0;
goto *0x7E00;`

I am attemping to jump to the bootloader to load a sketch, but something is wrong.  The following sketch doesn't work when running the bootloader mentioned above.  I compiled it countless times, and even changed the LED count to ensure I was loading the right bootloader.
`int counter=0;

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin 13 as an output.
  pinMode(13, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
  
  counter++;
  if (counter>=5) {
    cli();
    MCUSR=0;
    SP=RAMEND;
    goto *0x7E00;
  }
}`

It appears to get stuck jumping to the bootloader, the LED is blinking very fast.  An external reset enters the sketch when in this state, and the only way to enter the bootloader to load a sketch is by a 2nd external reset, while the application is running.

any ideas?  thanks -Mike

MarkG55, 2016-11-29:
@miketedeschi 
> I am attemping to jump to the bootloader to load a sketch, but something is wrong. 

This might seem like a silly question, but which controller are you testing this on?  Different controllers have different amounts of flash, so as long as I'm not mistaken, the address that you need to jump to will be dependant on the the controller that you're using.

miketedeschi, 2016-11-29:
@MarkG55 hi, i tried it on a 328p and a 328pb, same results.  End goal is RS485 uploads (which works with manual reset) but entering the bootloader from software is the snag.  I am just running the blink sketch + the suggested code running after 5 main loops, as shown above.

miketedeschi, 2016-11-29:
Also, i'm asuming if no sketch is uploaded, the above code should re-enter the blink application, blink 5 times, and enter the bootloader again.  right?

MarkG55, 2016-11-29:
Just looking at my code, I haven't got it clearing the interrupts flag - perhaps try commenting out the cli(); line and see if that makes any difference?

I would have thought that your code would work as it is, so I'm just guessing here...

miketedeschi, 2016-11-29:
hmm, same result with cli(); commented.

I also just tried moving the blink led to pin9 with another led so i can see if the bootloader or the sketch is blinking or the sketch is.  The code runs normally first pass at normal speed, then the bootloader blinks super fast, then the sketch blinks pin9 led forever very fast, so it's not stuck in the bootloader?

I also tried 2 leds alternating, it works through the first loop, then pin9 led blinks forever.  pin13 is excluded after the first run, which is ever stranger...

`int counter=0;

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin 13 as an output.
  pinMode(13, OUTPUT);
  pinMode(9, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(9, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(9, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000); 
  
  counter++;
  if (counter>=5) {
    cli();
    MCUSR=0;
    SP=RAMEND;
    goto *0x7E00;
  }
}`

miketedeschi, 2016-11-29:
and just in case, this is what i compiled to:
`:107E000001C0ACC0112484B7882349F0982F9A7020
:107E1000923029F081FF02C097EF94BFD7D085E060
:107E20008093810082E08093C00088E18093C1004C
:107E300086E08093C20080E18093C4008FE0B1D0DF
:107E4000259A86E020E33CEF91E030938500209373
:107E5000840096BBB09BFECF1D9AA8958150A9F7D0
:107E6000C0E0D0E092D0813471F48FD0182F9FD031
:107E7000123811F482E005C0113811F486E001C017
:107E800083E07CD068C0823411F484E103C085347F
:107E900019F485E094D05FC0853541F476D0C82FC1
:107EA00074D0D0E0D82BCC0FDD1F54C0863521F420
:107EB00084E085D080E0E5CF843681F566D065D05A
:107EC000D82E63D0F82E00E011E05FD0F801819346
:107ED0008F01DE12FACF6BD0F5E4FF1201C0FFCFA5
:107EE00040E050E063E0CE0139D07E0100E011E0D7
:107EF000F80140810E5F1F4F3196808150E0582B72
:107F000061E0C7012BD0F2E0EF0EF11CD012F0CFF0
:107F100040E050E065E0CE0121D01DC0843769F417
:107F200034D033D0182F31D042D07E01F701859163
:107F30007F0124D01150D1F70EC0853739F437D0E6
:107F40008EE11CD085E91AD08FE09BCF813511F4EA
:107F500088E027D02CD080E111D084CFFC010A0129
:107F600067BFE895112407B600FCFDCF667029F0C5
:107F7000452B19F481E187BFE89508959091C000E1
:107F800095FFFCCF8093C60008958091C00087FFC5
:107F9000FCCF8091C00084FD01C0A8958091C600EF
:107FA0000895E0E6F0E098E1908380830895EDDFA6
:107FB000803219F088E0F5DFFFCF84E1DFCFCF9387
:107FC000C82FE3DFC150E9F7CF91F1CF282E80E031
:087FD000E8DFE0E0FF2709945F
:027FFE00020679
:0400000300007E007B
:00000001FF
`

MarkG55, 2016-11-29:
It sounds like the controller is resetting constantly after the code tries to call the bootloader.

I'm assuming that's the complete code?  I've seen problems with using the watchdog cause constant restarts after a restart...

MarkG55, 2016-11-29:
I don't know if this will make any difference but perhaps try adding the following line above main():

void* BootLoader = (void *) 0x7E00;

Then change the goto line to the following:

      goto *BootLoader;

See if that makes any difference?

miketedeschi, 2016-11-29:
no change.  yea, thats all the code, i literally took the blink sketch and put the suggested code in to eliminate anything else like the bootloader.

I would have assumed it's resetting too.  Note that whatever LED is first in the sketch is the one that blinks forever, which would suggest the watchdog is only letting the sketch get far enough to blink the first led?

miketedeschi, 2016-11-29:
ill try disabling the watchdog in the setup, hold on

miketedeschi, 2016-11-29:
disabling watchdog did nothing, but see 3rd line in main loop.  if i comment that out, only pin9 blinks, but when UNcommented they both blink forever, so it's gotta be hitting that 3rd line... continuously.

`#include <Arduino.h>
#include <avr/wdt.h>
int counter=0;
void* BootLoader = (void *) 0x7E00;

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin 13 as an output.
  wdt_disable();
  pinMode(13, OUTPUT);
  pinMode(9, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  wdt_disable();
  digitalWrite(9, HIGH);   
  digitalWrite(13, HIGH); //BOTH BLINK unless this is commened out!
  delay(100);             
  digitalWrite(9, LOW);    
  delay(100);              
  digitalWrite(13, HIGH);   
  delay(100);              
  digitalWrite(13, LOW);  
  delay(100); 
  
  counter++;
  if (counter>=5) {
    cli();
    MCUSR=0;
    SP=RAMEND;
    goto *BootLoader;
  }
}`

MarkG55, 2016-11-29:
OK, so it's resetting when it hits the first delay statement.  What happens if you change the first delay to just a 1ms delay?

MarkG55, 2016-11-29:
Also, try commenting out SP=RAMEND and see if that changes anything?

miketedeschi, 2016-11-29:
//SP=RAMEND  no change

1ms was too fast to see what's going on, so i commened out p13 led.  now it's going back into the bootloader.  I see the triple bootloader blink (abnormally fast blinks), the 9p led gets stuck on for about 2 seconds.  but returning to the bootloader is new...

`void loop() {
  wdt_disable();
  digitalWrite(9, HIGH);   
  //digitalWrite(13, HIGH); //BOTH BLINK unless this is commened out!
  delay(1);             
  //digitalWrite(9, LOW);    
  delay(1);              
  //digitalWrite(13, HIGH);   
  delay(1);              
  //digitalWrite(13, LOW);  
  delay(1);

miketedeschi, 2016-11-29:
this change shuts p9 off before resetting...

`void loop() {
  wdt_disable();
  digitalWrite(9, HIGH);   
  //digitalWrite(13, HIGH); //BOTH BLINK unless this is commened out!
  delay(1);             
  digitalWrite(9, LOW);

MarkG55, 2016-11-29:
I was meaning for you to only change the **_first_** delay.

Have you tried removing SP=RAMEND?

miketedeschi, 2016-11-29:
yes.  i changed the other delays back, this blinks p9 (dim) forever, and p13 never comes back with the bootloader.

`// the loop function runs over and over again forever
void loop() {
  wdt_disable();
  digitalWrite(9, HIGH);   
  //digitalWrite(13, HIGH); //BOTH BLINK unless this is commened out!
  delay(1);             
  digitalWrite(9, LOW);    
  delay(100);              
  digitalWrite(13, HIGH);   
  delay(100);              
  digitalWrite(13, LOW);  
  delay(100); 
  
  counter++;
  if (counter>=5) {
    cli();
    MCUSR=0;
    //SP=RAMEND;
    goto *BootLoader;
  }
`}``

MarkG55, 2016-11-29:
OK, I'm really not sure what's going wrong for you.

There doesn't seem to be any easy way to contact you directly - if you can provide me with your email address, I'll send you .hex copy of my bootloader for you to try with and make sure that you're testing with exactly the same bootloader that I've got.

If you don't have any luck with it, I might try pulling out some hardware and test it for myself...

miketedeschi, 2016-11-29:
watchdog >> largewire.com would work, that would be amazing.  thank you

MarkG55, 2016-11-29:
OK, I've just sent you an email!