Optiboot/optiboot #141: auto reset not working on 3.3V, 8MHz Pro-Mini

bradselb, 2015-06-03:
Background: 
   - Sparkfun Arduino 3.3v, 8MHz Pro-Mini
   - Sparkfun Pocket AVR Programmer
   - current (Feb 2015) Adafruit FTDI cable (the one with the clear shell and blinky lights)
   - older (black shell) Adafruit "USB-Serial Debug Console cable" with Prolific PL2303HX chip
   - Arduino GUI ver 1.0.1 running on Debian 7
   - proto board and hook up wires. 

Steps to reproduce:
   - use proto board and jumpers to make an adapter from pocket programmer to pro-mini SPI bus.
   - Install Optiboot on the pro-mini using the pocket programmer and the Arduino GUI
   - disconnect the pro-mini from pocket programmer
   - connect the FTDI cable
   - download the "blink" example in the Arduino GUI.
   - observe that Avrdude fails (configure Arduino to be verbose about download).
   - even an appropriate command line invocation of avrdude fails.

Notes:
0. Sample size == 1 (!).   I have two more 3.3V, 8MHz pro-mini boards on order and can update with more info if desired. 

1. Following same steps and using the same equipment, Optiboot works well on the 5V, 16MHz pro-mini as well as several 5V, 16MHz nano boards. 

2. Programming Optiboot on the pro-mini appears to work fine. Avrdude completes successfully and...

3. Download of a program can be accomplished using the "USB-Serial Debug Console cable" using Avrdude on the command line and manipulating the reset button manually. Notice that this cable only connects TxD, RxD, Vcc and Gnd. This observation suggests that there may be a reset timing or duration issue (when the reset line is manipulated auto-magically by the FTDI cable)? 

3. Some more notes can be found in the comments of this makefile:  
    https://github.com/bradselb/avr/blob/master/blink/Makefile

WestfW, 2015-06-06:
Which version of the arduino IDE did you use to load the bootloader, and which bootloader (and version) did it burn?  Can you attach "verbose" logs of both the burn and the upload attempt?  Some of the optiboot targets are pretty old and may not quite match current shipping hardware.
This is probably NOT an auto-reset problem, since that happens in hardware.  All sorts of other things can go wrong and prevent a sketch upload from working.

bradselb, 2015-06-06:
I used Version 1.0.1 of the Arduino GUI environment.
Here is a transcript (copied from Arduino GUI) of burning the ne bootloader to the Sparkfun 3.3V, 8MHz Pro Mini...it looks successful to me.

```
/usr/share/arduino/hardware/tools/avrdude -C/usr/share/arduino/hardware/tools/avrdude.conf -v -v -v -v -patmega328p -cusbtiny -e -Ulock:w:0x3F:m -Uefuse:w:0x05:m -Uhfuse:w:0xdc:m -Ulfuse:w:0xff:m 

avrdude: Version 5.11.1, compiled on Oct 18 2011 at 21:41:15
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/usr/share/arduino/hardware/tools/avrdude.conf"
         User configuration file is "/home/brad/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/parport0
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 006:076
         AVR Part                      : ATMEGA328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported
[... 206 lines cut ...]
avrdude: verifying ...
avrdude: 32768 bytes of flash verified
avrdude: reading input file "0x0F"
avrdude: writing lock (1 bytes):

Writing | CMD: [58 00 00 00] [06 58 00 ff]
CMD: [ac e0 00 cf] [00 ac e0 00]
CMD: [58 00 00 00] [cf 58 00 ff]
CMD: [58 00 00 00] [00 58 00 cf]
################################################## | 100% 0.02s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0F:
avrdude: load data lock data from input file 0x0F:
avrdude: input file 0x0F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | CMD: [58 00 00 00] [00 58 00 cf]
################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.
```

bradselb, 2015-06-06:
Here is the verbose output of just trying to get avrdude to talk to the arduino via FTDI cable after installing Optiboot and typically, just prior to attempting download....

```
$ make 
avr-gcc -mmcu=atmega328p -DF_CPU=8000000 -Os -I.  -std=c99 -Wall -Wstrict-prototypes -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -Wl,--gc-sections -Wl,--relax -c -o blink.o blink.c
avr-gcc -mmcu=atmega328p -DF_CPU=8000000 -Os -I.  -std=c99 -Wall -Wstrict-prototypes -funsigned-char -funsigned-bitfields -ffunction-sections -fdata-sections -Wl,--gc-sections -Wl,--relax -o blink.elf blink.o
avr-objcopy -R .eeprom -O ihex blink.elf blink.hex
rm blink.o
$ make avrdude-check 
avrdude -p atmega328p -c arduino -P /dev/ttyUSB0 -b 115200 -v -v -v -v -v

avrdude: Version 5.11.1, compiled on Oct 18 2011 at 21:41:15
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/brad/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding

avrdude done.  Thank you.

make: *** [avrdude-check] Error 1
$ make avrdude-check 
avrdude -p atmega328p -c arduino -P /dev/ttyUSB0 -b 115200 -v -v -v -v -v

avrdude: Version 5.11.1, compiled on Oct 18 2011 at 21:41:15
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/brad/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding

avrdude done.  Thank you.

make: *** [avrdude-check] Error 1
$ make avrdude-check 
avrdude -p atmega328p -c arduino -P /dev/ttyUSB0 -b 115200 -v -v -v -v -v

avrdude: Version 5.11.1, compiled on Oct 18 2011 at 21:41:15
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/brad/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding

avrdude done.  Thank you.

make: *** [avrdude-check] Error 1
$ 

```

bradselb, 2015-06-06:
Here is a similar output but this time, using the Prolific PL2303 based USB-Serial cable and manipulating reset manually....

```
$ avrdude -p atmega328p -c arduino -P /dev/ttyUSB0 -b115200 -v -v -v -v

avrdude: Version 5.11.1, compiled on Oct 18 2011 at 21:41:15
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/brad/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: Recv: . [14] 
avrdude: Recv: . [10] 
         AVR Part                      : ATMEGA328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
[... 168 lines cut ...]
avrdude: Recv: . [14] 
avrdude: Recv: . [00] 
avrdude: Recv: . [10] 
avrdude: safemode read 1, efuse value: 0
avrdude: Send: V [56] P [50] . [08] . [00] . [00]   [20] 
avrdude: Recv: . [14] 
avrdude: Recv: . [00] 
avrdude: Recv: . [10] 
avrdude: safemode read 2, efuse value: 0
avrdude: Send: V [56] P [50] . [08] . [00] . [00]   [20] 
avrdude: Recv: . [14] 
avrdude: Recv: . [00] 
avrdude: Recv: . [10] 
avrdude: safemode read 3, efuse value: 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK
avrdude: Send: Q [51]   [20] 
avrdude: Recv: . [14] 
avrdude: Recv: . [10] 

avrdude done.  Thank you.

$ 

```

WestfW, 2015-06-10:
The 115200bps on 8MHz platforms is not very accurate.  It could be that the FTDI and 328 are off just enough that it doesn't work.  I usually recommend using the 16MHz bootloader and use 57600 upload speed for 8MHz platforms.  (Odd that it works with the prolific cable, though.)

fleinze, 2015-07-15:
I have the same issue. 
I am using Arduino 1.6.5 on Mac OS X 10.10, pro mini 3,3V/8MHz clones and a FTDI Tool for code Upload.

Steps to reproduce:
- Install Optiboot using the Board Manager
- Select Board: "Optiboot on 32-pin CPUs"
- Select Processor/CPU-Speed "Atmega328p" and "16MHz"
- Uploade Optiboot to a pro mini clone 8MHz via an Arduino as ISP. (successfull)
- Select CPU-Speed "8MHz (int)"
- Upload the "Blink" example

I got it to upload the sketch with manual reset, but it is quite hard to find the right moment for the reset.

fleinze, 2015-07-15:
Here ist what the avr-dude says:

    avrdude: Version 6.0.1, compiled on Apr 14 2015 at 16:30:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/florian/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-A602Z65Z
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
    avrdude: stk500_recv(): programmer is not responding
    Probleme beim Hochladen auf die Platine. Hilfestellung dazu unter http://www.arduino.cc/en/Guide/Troubleshooting#upload .

fleinze, 2015-07-15:
With manual Reset it looks like this:

    avrdude: Version 6.0.1, compiled on Apr 14 2015 at 16:30:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/florian/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-A602Z65Z
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
    avrdude: stk500_recv(): programmer is not responding
    avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 6.2
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

    avrdude: AVR device initialized and ready to accept instructions

    Reading | ################################################## | 100% 0.00s

    avrdude: Device signature = 0x1e950f
    avrdude: reading input file "/var/folders/27/14l23bd51gd0p9gphrsrhr6c0000gn/T/build1810089206869648059.tmp/Blink.cpp.hex"
    avrdude: writing flash (1030 bytes):

    Writing | ################################################## | 100% 0.57s

    avrdude: 1030 bytes of flash written
    avrdude: verifying flash memory against /var/folders/27/14l23bd51gd0p9gphrsrhr6c0000gn/T/build1810089206869648059.tmp/Blink.cpp.hex:
    avrdude: load data flash data from input file /var/folders/27/14l23bd51gd0p9gphrsrhr6c0000gn/T/build1810089206869648059.tmp/Blink.cpp.hex:
    avrdude: input file /var/folders/27/14l23bd51gd0p9gphrsrhr6c0000gn/T/build1810089206869648059.tmp/Blink.cpp.hex contains 1030 bytes
    avrdude: reading on-chip flash data:

    Reading | ################################################## | 100% 0.48s

    avrdude: verifying ...
    avrdude: 1030 bytes of flash verified

    avrdude done.  Thank you.

    Probleme beim Hochladen auf die Platine. Hilfestellung dazu unter http://www.arduino.cc/en/Guide/Troubleshooting#upload .

closed 2016-12-19 by WestfW