Optiboot/optiboot #146: optiboot for atmega128

vanbwodonk, 2015-06-29:
Sir,

My name is Arif, i would like to edit optiboot to fit into atmega128.
In my mind atmega128 have same flash and ram with atmega1280 then i think this will be at least same with atmega128.

i change optiboot to this,
in pin_defs.h

/*------------------------------------------------------------------------ */
/* Mega support */
#if defined(__AVR_ATmega128__)
/*------------------------------------------------------------------------ */
/* Onboard LED is connected to pin PB7 on Arduino Mega */ 
#if !defined(LED)
#define LED         B7
#endif
#define TIFR1 	TIFR
#define WDTCSR	WDTCR

in optiboot.c

#elif defined(__AVR_ATmega128__)
#undef RAMSTART
#define RAMSTART (0x200)
#endif

in the makefile

#Atmega128
atmega128: MCU_TARGET = atmega128
atmega128: CFLAGS += $(COMMON_OPTIONS) -DBIGBOOT $(UART_CMD)
atmega128: AVR_FREQ ?= 16000000L
atmega128: LDSECTIONS  = -Wl,--section-start=.text=0x1fc00  -Wl,--section-start=.version=0x1fffe
atmega128: $(PROGRAM)_atmega128.hex
atmega128: $(PROGRAM)_atmega128.lst

and after compiled i got this error,

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Error=-2.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega128 -DF_CPU=16000000L  -DBAUD_RATE=11
5200 -DLED_START_FLASHES=3      -DBIGBOOT    -c -o optiboot.o optiboot.c
optiboot.c:302:2: warning: #warning BAUD_RATE error greater than -2%
C:\Users\ELKA\AppData\Local\Temp/cca3wRWu.s: Assembler messages:
C:\Users\ELKA\AppData\Local\Temp/cca3wRWu.s:461: Error: number must be positive and less than 64
C:\Users\ELKA\AppData\Local\Temp/cca3wRWu.s:504: Error: number must be positive and less than 64
C:\Users\ELKA\AppData\Local\Temp/cca3wRWu.s:522: Error: number must be positive and less than 64
C:\Users\ELKA\AppData\Local\Temp/cca3wRWu.s:540: Error: number must be positive and less than 64
make: *** [optiboot.o] Error 1

any suggestions sir?

majekw, 2015-07-06:
You could try to add `-save-temps` to gcc options and check what is in cca3wRWu.s at indicated lines. I bet it's 'in' or 'out' instruction, but more context is needed to locate right code which generates it.

vanbwodonk, 2015-07-07:
You're correct sir, this inside .s in that line.
461 : out 72, r9
504 : out 72, r8
522 : out 72, r7
540 : out 72, r10
Any further suggestions?

majekw, 2015-07-07:
Could you give some contents of .s file around these lines?
For example something between 500 and 522?

vanbwodonk, 2015-07-08:
This 499 ~ 525,
```
.LSM60:
	movw r30,r20
/* #APP */
 ;  898 "optiboot.c" 1
	movw  r0, r24
	out 72, r8
	spm
	clr  r1
	
 ;  0 "" 2
.LSM61:
/* #NOAPP */
	subi r20,lo8(-(2))
	sbci r21,hi8(-(2))
.LBE31:
.LSM62:
	subi r22,lo8(-(-2))
	sbci r23,hi8(-(-2))
	brne .L42
.LSM63:
	movw r30,r12
/* #APP */
 ;  905 "optiboot.c" 1
	out 72, r7
	spm
	
 ;  0 "" 2 
```

majekw, 2015-07-08:
Atmega128 has SPMCSR register at 0x68, so it's inaccessible by `out` instruction.
You need to change `__boot_*_short` calls by `__boot_*_normal` and `boot_rww_enable` by `__boot_rww_enable`

vanbwodonk, 2015-07-09:
Thank your sir, my atmega128 can used optiboot with this change on optiboot.c
- no need to change RAMSTART (RAMSTART = 0x100)
- add defined (__AVR_ATmega128__) to all defined(__AVR_ATmega8__) || defined (__AVR_ATmega32__)
- change __boot_*_short calls by __boot_*_normal and boot_rww_enable by __boot_rww_enable
```
avrdude.exe: Version 6.1, compiled on Apr  8 2014 at 14:19:37
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "D:\5.Software\10.Electronic APP\AVRDUDESS\avrdudess_20140805\avrdude.conf"

             Using Port                    : COM14
             Using Programmer              : arduino
             Overriding Baud Rate          : 115200
             AVR Part                      : ATmega128
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
               flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
               lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

             Programmer Type : Arduino
             Description     : Arduino
             Hardware Version: 3
             Firmware Version: 6.2
             Vtarget         : 0.3 V
             Varef           : 0.3 V
             Oscillator      : 28.800 kHz
             SCK period      : 3.3 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9702
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file "C:\Users\ELKA\AppData\Local\Temp\build2108472660544326218.tmp/MultiSerialMega.cpp.hex"
avrdude.exe: input file C:\Users\ELKA\AppData\Local\Temp\build2108472660544326218.tmp/MultiSerialMega.cpp.hex auto detected as Intel Hex
avrdude.exe: writing flash (2346 bytes):

Writing | ################################################## | 100% 0.33s

avrdude.exe: 2346 bytes of flash written
avrdude.exe: verifying flash memory against C:\Users\ELKA\AppData\Local\Temp\build2108472660544326218.tmp/MultiSerialMega.cpp.hex:
avrdude.exe: load data flash data from input file C:\Users\ELKA\AppData\Local\Temp\build2108472660544326218.tmp/MultiSerialMega.cpp.hex:
avrdude.exe: input file C:\Users\ELKA\AppData\Local\Temp\build2108472660544326218.tmp/MultiSerialMega.cpp.hex auto detected as Intel Hex
avrdude.exe: input file C:\Users\ELKA\AppData\Local\Temp\build2108472660544326218.tmp/MultiSerialMega.cpp.hex contains 2346 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.24s

avrdude.exe: verifying ...
avrdude.exe: 2346 bytes of flash verified

avrdude.exe done.  Thank you.
```

hsaul17, 2016-04-21:
Hi  vanbwodonk, I just bought an atmega128, i'm trying to build the bootloader but i'm not good at that, can you share the one that you build?

thanks in advance

vanbwodonk, 2016-04-22:
https://github.com/vanbwodonk/optiboot128

WestfW, 2018-08-07:
Added as part of https://github.com/Optiboot/optiboot/commit/b263cada0830eb4ec7a91fbe0bb5a7b85f5b5166
Tested on a Wiring 1 board.

closed 2018-08-07 by WestfW