Optiboot/optiboot #150: Support Atmel Studio users

WestfW, 2015-07-21:
It would be nice if it was easier for Atmel Studio users to compile optiboot from source.
Things don't go well using the default AS Makefile :-(

WestfW, 2015-09-24:
I'm going to declare this "done", except for documentation.
https://github.com/Optiboot/optiboot/commit/fc2368bbf506e53c5ec7d302df7e275f0b267577

syddo, 2017-03-15:
Hi @WestfW I am unable to compile optiboot source in AS7 using the makefiles. :(

WestfW, 2017-03-17:
@syddo: Are you seeing https://github.com/Optiboot/optiboot/issues/200 or something else?

joelmuzz, 2017-07-14:
It is building for me (with atmega328) but the build reports as a failure due to "error: recipe for target 'baudcheck' failed". That may be the issue mentioned.
Actual hex seems ok.

[output.txt](https://github.com/Optiboot/optiboot/files/1147377/output.txt)