Optiboot/optiboot #154: Incorrect lock/unlock bit values in boards-1.6.txt?

per1234, 2015-09-09:
either optiboot28 or optiboot32 in boards-1.6.txt seems to have the values swapped.

from boards-1.6.txt:
```
optiboot28.bootloader.unlock_bits=0x0F
optiboot28.bootloader.lock_bits=0x2F
```
```
optiboot32.bootloader.unlock_bits=0x2F
optiboot32.bootloader.lock_bits=0x0F
```
 
Also, are the values intended to be different in boards.txt and boards-1.6.txt?

from boards.txt:
```
atmega328o.bootloader.unlock_bits=0x3F
atmega328o.bootloader.lock_bits=0x0F
```
from boards-1.6.txt:
```
optiboot1284.bootloader.unlock_bits=0x3F
optiboot1284.bootloader.lock_bits=0x2F
```
from boards.txt:
```
atmega1284o.bootloader.unlock_bits=0x3F
atmega1284o.bootloader.lock_bits=0x0F
```
etc.

WestfW, 2015-09-10:
0x2F allows the application to read the boot segment, which enables it to retrieve the bootloader version (for example.)   This is a change I made that I couldn't get the Arduino folks to accept, because "it wasn't that way before."

WestfW, 2015-09-10:
(I'll need to re-check the  "swapped" values.)

WestfW, 2016-02-06:
Fixed in commit https://github.com/Optiboot/optiboot/commit/be06597525886d9e6c36a0496f570d427b667e05

closed 2016-02-06 by WestfW