Optiboot/optiboot #156: Add support for Atmel Xplained Mini AVR Eval boards.

WestfW, 2015-09-24:
Atmel has three "Xplained mini" evaluation boards that are more-or-less arduino compatible:
Xplained Mini 168pb with Atmega168pb chip
Xplained Mini 328p with Atmega328p chip (just like an Uno)
Xplained Mini 328pb with Atmega328pb chip (adds significant additional features: more timers, 2nd serial port, 2nd I2C port, Touch Control peripheral, unique device id)
The bootloader support is relatively trivial; the boards include an Atmel "mEDBG" debug chip, which acts as a USB/Serial at up to 57600bps (there are problems with higher speeds.)  Currently avrdude does not support programming via the mEDBG, so you have to use Atmel Studio for the "burn bootloader" phase.  But once that is done, the chips will act as normal 168 or 328p chips.

So, several parts:
1) add "convenient" targets to the Makefiles.
2) adjust boards.txt
3) document

WestfW, 2015-09-24:
(1) done: https://github.com/Optiboot/optiboot/commit/56e2952e17314ad7aa23582441bbe53841527c1f

aethaniel, 2017-03-23:
@WestfW, AFAIK this is working perfectly since 2y. I have integrated the binaries in https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/releases.
This Arduino module is natively available from Arduino IDE Boards Manager since a while.

WestfW, 2017-03-24:
Still needs done:
 1) General Documentation.
 2) Find a way to load the bootloader on systems without AS tools.  Some combination of OpenOCD, third-party tools, or "newer" versions of AVRDude?

sijskes, 2017-12-11:
@WestfW firmware loaded here with usbtiny, on the ISP connector.

closed 2021-10-08 by WestfW