Optiboot/optiboot #162: Add support for ATmega8535

hansibull, 2015-10-30:
Could you please add support for the ATmega8535? I got a few of those lying around, and Iid love to be able to have a bootloader for them. Should be an easy port. It's almost identical to the ATmega16, except for JTAG and just 8kB of flash

WestfW, 2018-08-07:
https://github.com/Optiboot/optiboot/commit/b263cada0830eb4ec7a91fbe0bb5a7b85f5b5166
(not tested)

closed 2018-08-07 by WestfW

MCUdude, 2018-08-07:
FYI I've tested optiboot_flash on the following devices with success:

* ATmega8
* ATmega16
* ATmega32
* ATmega64
* ATmega88, P, PA, PB
* ATmega128
* ATmega162
* ATmega164A, P
* ATmega324A, P, PA, PB
* ATmega328, P, PB
* ATmega644A, P
* ATmega1281
* ATmega1284, P
* ATmega2560
* ATmega2561
* ATmega8515
* ATmega8535

BTW note that ATmega162 is different when it comes to UART initialization than any other AVR I've used.
I needed to add a [separate UART_SRC line](https://github.com/MCUdude/optiboot_flash/blob/eefd51b0f74d343d40d9028723fdb0dfb2f0a295/optiboot_flash.c#L560) just for this device.

WestfW, 2018-08-07:
Excellent.  I see I missed that m162 line...

WestfW, 2018-09-20:
Do you have any additional test programs for the application-flash-write feature (beyond the SerialReadWrite sketch)?

MCUdude, 2018-09-20:
No, I'm sorry. I made this sketch a few years ago just as a proof of concept. It would be awesome if a better sketch was made, or maybe a proper library. My dream is a library like the SD library with a simple file system. However I'm not capable of making something like this 😉

MCUdude, 2018-09-20:
FYI I've also been working on my own version of Optiboot in [this fork](https://github.com/MCUdude/optiboot_flash/tree/v7.1). The difference between this and the version you've added to release 8.0 are these:
* LD_SECTION is now changed if eeprom support is anabled for "smaller" devices. "eeprom_support" is also added to the file name
* Most of the "noisy" output from gcc is now muted. Now only errors will now be printed (apart from the usual stuff).