Optiboot/optiboot #164: Compatibility ARDUINO 1.6.6 IDE

sebilativi, 2015-11-21:
Hello, 
thanks for the great bootloader. 

I wish optiboot would be cleaned up and be compatible with the new 1.6.6 IDE. This means I just have to generate a directory named 'hardware' in my Sketchbook, extract the optiboot files and would be finished. 

With IDE 1.6.6 I had java and panic errors. Tried everything, but was not able to get it running. Downloaded IDE 1.6.5, changed to 'boards-1.6.txt', copied the contents in the IDE 'boards.txt', copied the bootloaders in the IDE, deleted the 'hardware' dir and now it's working... 

I would clean this all up by myself, but sorry, I was not able to find the problem with the IDE 1.6.6. Maybe you could look at it. 

Thanks and kindest regards, 
Sebilativi

majekw, 2015-11-21:
I use Optiboot with Arduino 1.6.6. It took me also a while to discover how it should be installed :-)
So, in sketch folder you need 'hardware' directory as you did first. Put Optiboot inside and rename second level directory 'optiboot' to 'avr'.
Directory tree should look like this:
```
hardware/
└── optiboot
    └── avr
        ├── bootloaders
        │   └── optiboot
        └── libraries
```

I think that release.sh script should be updated to create two different packages for each version family of Arduino. Best option would be to generate correct boards.txt from one source to not maintain two separate files as it's done currently.

WestfW, 2015-11-22:
Did you try it via the .json file on the "releases" page?  ("beta"...)  I didn't think that there was a change between 1.6.5 and 1.6.6...  (although, this is .hex files only.   The source tree hasn't been compatible with the IDE in a LONG time, and 1.6.x no longer includes the needed tools, anyway.  So I'm not sure what should be done there...)

2114L3, 2015-11-25:
atmega8 issues as well.

needs bootloader change as main platform.txt file now uses efuse which is not used for atmega8.
so the avr folder needs its own platform.txt file with "-Uefuse:w:{bootloader.extended_fuses}:m" removed in a new avrdude bootloader label eg; 
```
tools.avrdude_atm8.bootloader.pattern="{cmd.path}" "-C{config.path}" {bootloader.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{runtime.platform.path}/bootloaders/{bootloader.file}:i" -Ulock:w:{bootloader.lock_bits}:m
```
then in boards.txt
```
atmega8_opti_8mhz.bootloader.tool=arduino:avrdude_atm8
```

optiboot_atmega8.hex which is referenced in the boards-1.6.txt is also missing from the bootloader folder

as a results this zip is just unusable for atmega8 and was easier to hack an old 1.0.3 version that still had the hex

west727, 2016-09-25:
plese put this info in readme

WestfW, 2016-09-26:
I've updated the README file.
https://github.com/Optiboot/optiboot/commit/0eb5e3710a961754d978a5f86a8d3fbe1e588751
Still need to update the more detailed wiki pages...

Roberto6969, 2018-09-21:
I see this issue is still open so I will use it. Maybe this will help to clear some question regarding installation in IDE environment (mine is 1.8.7):
1. on location "C:\Users\test\AppData\Local\Arduino15\packages\Optiboot_old\hardware\avr" should be folder 0.8.0 as it was for version 0.6.2 and 0.7.0? 
2. Is @majekw advice still valid?  https://github.com/Optiboot/optiboot/issues/164#issuecomment-158685992
3. Should be any optiboot's folder here? C:\Program Files (x86)\Arduino\hardware\arduino\avr\bootloaders

I have made installation/deinstallation/installation but now Boards Manager show Optiboot 8.0 by westfw version 0.8.0 INSTALLED but ther is no traces of files on computer :-(

WestfW, 2018-09-21:
1) When I install Optiboot 8.0 on a windows system, I wind up with the actual .hex files in:
"C:\Users\billw\AppData\Local\Arduino15\packages\Optiboot\hardware\avr\0.8.0\bootloaders\optiboot\optiboot_atmega1280.hex"
**Except I seem to have left out at atmega328 .hex file** :-(

2) I *think* the current board-manager install currently puts things in the correct place; the burn-bootloader command does the right thing with my 1.8.5 install; I'll check with 1.8.7 as well...

3) No.  or rather, that's the original version of optiboot (~2011) that comes with the IDE.
  All the things installed by the board manager go in the "packages" directory.

Understand that "AppData" is a hidden, system file, so its contents do not show up in search results, and with a normal user-level configuration you can't even see it to look at.  And it doesn't install source, anyway.  Unless you're interested in only the pre-built binary for one of the "main" cores, using the board manager is a poor way to work with Optiboot.  See https://github.com/Optiboot/optiboot/issues/243

Roberto6969, 2018-09-22:
Thank you for your explanation. You have pointed out right source of possible problem. I have installed in 1.8.7 and folder was empty.
Then i installed in 1.8.5 version and got C:\Users\test\AppData\Local\Arduino15\packages\Optiboot\hardware\avr\0.8.0\bootloaders\optiboot six *.hex files including optiboot_atmega328.hex

It seems something goes wrong in version 1.8.7 during the installation because directory structure in folder packages is not created.