Optiboot/optiboot #165: "Release" Atmel Studio targets incorrect...

WestfW, 2015-11-25:
Apparently, the "release" targets of the Atmel Studio projects (https://github.com/Optiboot/optiboot/issues/150 ) are not correct (I only ever built "debug" targets.)

aethaniel, 2015-11-25:
I can confirm the "debug" works very well at the moment :wink:

syddo, 2017-03-15:
@aethaniel I can't compile in AS7... have you tried?

aethaniel, 2017-03-15:
I always used existing makefile with avr-gcc from studio in the path. Worked perfectly every time.

WestfW, 2017-03-17:
>> can't compile in AS7
Sheesh.   Do you get past the "upgrade" to the new AS infrastructure?  I can't even get the project to load under the latest AS7.   Grr.

aethaniel, 2017-03-17:
Oooops, i didn't see the history of the thread last time. Sorry.
I will take look during the week end to see if i can bring any help.

WestfW, 2017-03-22:
Is this https://github.com/Optiboot/optiboot/issues/200 ?
(which should be fixed now.)

WestfW, 2018-07-27:
https://github.com/Optiboot/optiboot/commit/70bd967c3cc5e242f09cb4849e2ce396bfc60fe7

closed 2018-07-27 by WestfW