Optiboot/optiboot #167: warning when compiling for atmega328

schmurfy, 2015-12-17:
I wanted to play with the bootloader mainly to learn how it works but I get a strange warning when trying to compile it:

```
BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Error=2.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3 -c -o optiboot.o optiboot.c
#warning BAUD_RATE error greater than 2%
```

I have not changed anything, just cloned the repository and ran `make atmega328`

I am using avr-gcc v4.8.3 on mac os x installed by homebrew

WestfW, 2015-12-18:
This is, alas, normal.  A 16MHz AVR does not do 115200bps very accurately.  There should be a total error margin of 5% allowable, so 2.1% is within limits (a warning rather than an error.)

closed 2015-12-18 by WestfW

schmurfy, 2015-12-19:
thanks, I thought I did something wrong.