Optiboot/optiboot #168: Add 8MHz external crystal for 328P (as in the Pro Mini 3.3V)

fra87, 2015-12-28:
Hello
It would be great if you add the support to the 8MHz external crystal for, at least, the 328P. Why, if you just have the 8MHz internal clock? Because boards like the Pro Mini 3.3V have a 8MHz external crystal...

You just have to add these lines

optiboot28.menu.mhz.8MHzExt=8MHz (ext)
optiboot28.menu.mhz.8MHzExt.build.f_cpu=8000000L
optiboot28.menu.mhz.8MHzExt.bootloader.low_fuses=0xFF
optiboot28.menu.mhz.8MHzExt.upload.speed=57600

in the proper section of the boards.txt file

musgravehill, 2016-01-26:
Hello, I made the same fix.
But I can not decide, what bootloader to use: 
optiboot_atmega328.hex (like in boards-1.6.txt for Pro Mini) 
OR optiboot_atmega328_pro_8MHz.hex (like in boards.txt)

fra87, 2016-01-26:
If I'm not mistaken I flashed the optiboot_atmega328.hex, since I used boards-1.6.txt. I downloaded everything using the xml file, so that one was the included bootloader.

musgravehill, 2016-01-26:
Thanks!
I created a special type for Pro Mini 3.3V 8Mhz [external crystal, BOD 1.8++ V]

optibootlowpower33.name=Optiboot LowPower 1.8++V 8MHz

optibootlowpower33.build.board=AVR_UNO
optibootlowpower33.build.core=arduino:arduino
optibootlowpower33.build.variant=arduino:eightanaloginputs
optibootlowpower33.build.f_cpu=8000000L
optibootlowpower33.build.mcu=atmega328p

optibootlowpower33.upload.tool=arduino:avrdude
optibootlowpower33.upload.protocol=arduino
optibootlowpower33.upload.speed=57600
optibootlowpower33.upload.maximum_size=32256
optibootlowpower33.upload.maximum_data_size=2048

optibootlowpower33.bootloader.tool=avrdude
optibootlowpower33.bootloader.unlock_bits=0x2F
optibootlowpower33.bootloader.lock_bits=0x0F
optibootlowpower33.bootloader.file=optiboot/optiboot_atmega328_pro_8MHz.hex

optibootlowpower33.bootloader.low_fuses=0xFF
optibootlowpower33.bootloader.high_fuses=0xDE
optibootlowpower33.bootloader.extended_fuses=0x06

-----
But this does not work. I have to find a reason... 
I`m not sure about: 57600, 8000000L, lock/unlock bits and fuses L,H,E.
And "optiboot_atmega328_pro_8MHz.hex" VS "optiboot_atmega328.hex"

fra87, 2016-01-26:
Are you on arduino <1.6? If not, you should mimic the boards-1.6.txt file, not the boards.txt. You will see that the syntax is quite different from yours (which is the older one)
If you are using the older IDE, anyway, try starting from a working configuration and then modify it.

Also.. check the baud rate; The "optiboot_atmega328.hex" was compiled for a 16MHz atmega328p and 115200 b/s, so you specify 57600 as baud rate (it thinks he's going twice the speed, but it is not, so your pc has to slow down). The "optiboot_atmega328_pro_8MHz.hex", on the other side, COULD need 115200. Verify with a working configuration

musgravehill, 2016-01-26:
I`m on IDE 1.6.6 from arduino.cc
In the boards-1.6.txt for all Pro Mini (32 pin) use "optiboot_atmega328.hex" and 57600 baudrate.

musgravehill, 2016-01-26:
I can not run optiboot on ATmega 328p with External oscillator 8MHz 3.3V.
So, I compile scetch on Arduino IDE with wdt, etc. Then push *.hex to clear ATmega without any bootloaders. It works fine!

fra87, 2016-01-26:
Wait, I used optiboot with arduino 1.6.6 and atmega328p with external oscillator 8MHz 3.3V, so I'm sure you can.
This is how I did it.
1) download file from https://github.com/Optiboot/optiboot/releases/download/v6.2/optiboot.zip
2) extract it inside your sketch folder so that the boards.txt file has the following path
      sketchFolder\hardware\optiboot\avr\boards.txt
    of course you should keep the folder structure (so the bootloaders hex files will be in folder sketchFolder\hardware\optiboot\avr\bootloaders\optiboot, for instance)
3) edit the boards.txt and add the four lines from the first post (JUST the four lines of the first post). I added them under line 38 (which was optiboot28.menu.mhz.8MHz.upload.speed=57600)

You are done. After this launch arduino ide, flash the bootloader with an external programmer, connect to the serial port and yiou can upload the sketches

musgravehill, 2016-01-27:
What about:  
"optiboot28.menu.mhz.8MHzExt.upload.speed=57600"  for ProMini should be "optiboot32".
 
The 28-pin and 32-pin CPUs uses "optiboot_atmega328.hex" with various upload speed: 57600 and 115200. So, is "optiboot_atmega328.hex" support various baudrate?

fra87, 2016-01-27:
Yeah, sorry, I was using a 28 pin atmega.
Anyway no, as far as I know the 57600 devices are the 8MHz ones, while the 115200 are the 16MHz ones. This is because the bootloader is compiled for 115200@16MHz, so when running at 8MHz the actual baud rate it is using is 57600. So when communicating with a 8MHz bootloader you can either
1) enter your PC's processor and set its clock speed to half the nominal one, like the atmega one
2) enter a baud rate of 57600
the most easy-to-do solution is, of course, number 2 ;)

musgravehill, 2016-01-28:
Ha-ha, I agree with you! Thanks for good idea about upload speed reduction on 8Mhz.

aster1cs, 2017-10-02:
I too just tried for the first time to bootload a 328p (8MHz, 3.3V), but it took me quite some time to figure out I'd have to change the BAUD rate in the boards.txt from 115200 to 57600 (using [Optiboot] Arduino Pro or Pro Mini (3.3V, 8 MHz) w/ ATmega328). It looks like I'm not alone with this issue. Is there a chance it will be changed? ;-)

Phoenix1747, 2018-06-12:
@fra87 So just to re-cap here, to use a 8mhz external crystal for the atmega328p you only need to flash the standard optiboot bootloader and then use the correct settings from the boards.txt (e.g. for Arduino Pro or Pro Mini (3.3V, 8 MHz) w/ ATmega328P or Adafruit Feather 328P) to push the sketches from the Arduino IDE?

fra87, 2018-06-15:
@Phoenix1747 Uh, it's been quite a long time since I did this. In any case, the process depends on which version of arduino ide you are using.

- if you are using an older version (<1.6, which uses boards.txt) I don't think you need much to do. Select that variant from the menu, then flash the bootloader (it will flash optiboot_atmega328_pro_8MHz.hex).
- if you are using a newer version (>=1.6, which uses boards-1.6.txt) you have to modify the file to add the feature.

If you have arduino 1.6 or greater, it MAY be possible that your boards file is called boards.txt, even if it is the boards-1.6.txt file. Just compare it with the [boards.txt file](https://github.com/Optiboot/optiboot/blob/master/optiboot/boards.txt) or the [boards-1.6.txt file](https://github.com/Optiboot/optiboot/blob/master/optiboot/boards-1.6.txt) in the repository.

In order to do the modification, open the file and go to line 43. As you can see, the section has the comment

> For 8MHz using the internal RC Oscillator, we adjust fuses, use the same bootloader binary, and halve the upload rate.

So, at line 43 paste the following code:

```
optiboot28.menu.mhz.8MHzExt=8MHz (ext)
optiboot28.menu.mhz.8MHzExt.build.f_cpu=8000000L
optiboot28.menu.mhz.8MHzExt.bootloader.low_fuses=0xFF
optiboot28.menu.mhz.8MHzExt.upload.speed=57600
```

You will now have the possibility to select "8MHz (ext)" as speed

Phoenix1747, 2018-06-15:
Ok thanks!

Shouldn't it work though if I just use a custom configuration like this:
```
custom328p.name=Custom 328P Board
custom328p.upload.tool=arduino:avrdude
custom328p.upload.protocol=arduino
custom328p.upload.maximum_size=32256
custom328p.upload.maximum_data_size=2048
custom328p.upload.speed=57600
custom328p.bootloader.tool=arduino:avrdude
custom328p.bootloader.low_fuses=0xFF
custom328p.bootloader.high_fuses=0xDE
custom328p.bootloader.extended_fuses=0x05
custom328p.bootloader.unlock_bits=0x3F
custom328p.bootloader.lock_bits=0x0F
custom328p.bootloader.file=optiboot/optiboot_atmega328.hex
custom328p.build.mcu=atmega328p
custom328p.build.f_cpu=8000000L
custom328p.build.core=arduino:arduino
custom328p.build.variant=arduino:standard
custom328p.build.board=AVR_PRO
```

or in addition I could try to use the Arduino Pro Mini 3.3V (8MHz) board from the standard Arduino boards.txt, right?

```
pro.name=Arduino Pro or Pro Mini

pro.upload.tool=avrdude
pro.upload.protocol=arduino

pro.bootloader.tool=avrdude
pro.bootloader.unlock_bits=0x3F
pro.bootloader.lock_bits=0x0F

pro.build.board=AVR_PRO
pro.build.core=arduino
pro.build.variant=eightanaloginputs

## Arduino Pro or Pro Mini (3.3V, 8 MHz) w/ ATmega328P
## ---------------------------------------------------
pro.menu.cpu.8MHzatmega328=ATmega328P (3.3V, 8 MHz)

pro.menu.cpu.8MHzatmega328.upload.maximum_size=30720
pro.menu.cpu.8MHzatmega328.upload.maximum_data_size=2048
pro.menu.cpu.8MHzatmega328.upload.speed=57600

pro.menu.cpu.8MHzatmega328.bootloader.low_fuses=0xFF
pro.menu.cpu.8MHzatmega328.bootloader.high_fuses=0xDA
pro.menu.cpu.8MHzatmega328.bootloader.extended_fuses=0xFD
pro.menu.cpu.8MHzatmega328.bootloader.file=atmega/ATmegaBOOT_168_atmega328_pro_8MHz.hex

pro.menu.cpu.8MHzatmega328.build.mcu=atmega328p
pro.menu.cpu.8MHzatmega328.build.f_cpu=8000000L
```
the IDE will then flash ATmegaBOOT, not optiboot obviously, but if it works :D

fra87, 2018-06-15:
IDE version?

closed 2018-06-15 by fra87