Optiboot/optiboot #171: Wrong lock/unlock bits

fra87, 2015-12-29:
Hello
Just a short info on lock and unlock bits in the boards.txt file. There you can read

optiboot28.bootloader.unlock_bits=0x0F
optiboot28.bootloader.lock_bits=0x2F

and, then, 

optiboot32.bootloader.unlock_bits=0x2F
optiboot32.bootloader.lock_bits=0x0F

Now, I'm not really into lock bits, but I tried to read the datasheet and I found that (page 281 of the complete datasheet, Table 28-3) I think that they should BOTH be 

optibootXX.bootloader.unlock_bits=0x3F
optibootXX.bootloader.lock_bits=0x0F

This is also confirmed by the original arduino boards.txt file, where for both the arduino UNO (28-pin 328p) and the pro mini (32-pin 328p) these values are used.

Now, I think that these are useless values, since once programmed the bits cannot be reset (so once locked you can't unlock unless you completely erase the chip), but... This should be fixed ;)

WestfW, 2016-02-02:
Use of 0x2F for "lock bits" instead of 0x0F (as in the original Arduino boards.txt) is intentional, and allows the sketch to read the boot area (in order to determine bootloader version, for instance.)

Use of 0x0F instead of 0x3F for "unlock bits" is a mistake.

fra87, 2016-02-02:
Ok, so they should be

optiboot28.bootloader.unlock_bits=0x3F
optiboot28.bootloader.lock_bits=0x2F
and
optiboot32.bootloader.unlock_bits=0x3F
optiboot32.bootloader.lock_bits=0x2F

right?

WestfW, 2016-02-06:
Fixed in commit https://github.com/Optiboot/optiboot/commit/be06597525886d9e6c36a0496f570d427b667e05

closed 2016-02-06 by WestfW