Optiboot/optiboot #172: compiling optiboot for: atmega1284 UART=1 AVR_FREQ=8000000L

github247, 2016-01-12:
Hi all,
I want to use an atmega1284p with visualmicro. Therefore I'd like to use optiboot as bootloader. My problem is, I can't access Serial0. I must use Serial1.

As far as I read, I need to compile a new Bootloader hex-file. But I don't know how to do. I'm using Windows and arduino IDE 1.6.4. There should be everything in, but I can't find it.

I tryed to find make.exe, but it seems like it's not on my system. Can you maybe explain for a dummy how to compile?

I guess it's: 
make UART=1 AVR_FREQ=8000000L atmega1284 LED_START_FLASHES=0

But how can I access make.exe and where is the hex file saved? 

By the way. once I've got the hex, I just need to flash this with my AVR ISP into the atmel or?

Many thanks!

BR Chris

WestfW, 2016-02-02:
You need a command-line AVR development environment (including make.exe) to compile Optiboot.  This used to be part of the Arduino distribution, and the easiest way to set it up on windows is probably to install the Arduino IDE 1.0.6, put the github Optiboot source parallel to the existing .../hardware/arduino/bootloaders/optiboot directory, and use the "omake" .bat file.  You probably should also use 57600bps for an 8MHz chip:
~~~
   omake UART=1 AVR_FREQ=8000000L LED_START_FLASHES=0 BAUD_RATE=57600 atmega1284
~~~

closed 2016-02-02 by WestfW

trivalik, 2017-03-05:
BTW: In current Arduino IDE 1.8.0 is no make tool inside anymore.