Optiboot/optiboot #177: Link troubles with Atmel toolset (avr-gcc 4.9.x)

WestfW, 2016-02-06:
The structure of the avr-libc libraries has apparently changed in the atmel toolchain containing 4.9.x, and the "-lc" at the end of the link command no longer allows optiboot builds with eeprom support to find the eeprom_xxx_byte() functions (they moved from lib/avrN/libc.a to lib/avrN/lib<chipname>.a)

stefanrueger, 2016-07-26:
For 4.9.x you need to specify -l$(MCU_TARGET) in the Makefile (as opposed to -lc). The new avr-gcc provides chip-specific libraries, for example, see  /usr/lib/avr/lib/avr5/libatmega328p.a. I find that 4.9.2 often creates longer code than 4.8.1.

I have rewritten optiboot to get rid of any library calls, so I can place the address variable in a global call-clobbered register Z=r30:r31, which makes the code shorter. I needed the space for a dualboot capability (check on WDT reset whether there is a bootable image on external SPI memory and burn that, so that I can do "over the air" programming for projects that have a radio/wifi). I managed to shorten the standard optiboot to 326 bytes (atmega328p), so had ample space for dual-boot, proper eeprom r/w support, for a pgm_write_page() routine that can be called from the application to write to PROGMEM (and use that like EEPROM) and for code that ensures that optiboot won't overwrite itself. All of that within 512 bytes (atmega328p, but only with 4.8.1; with the newer compiler I need to drop one feature to stay within 512 bytes). The code now compiles for > 110 chips, but I only tested 6: attiny167, attiny85, atmega1284p, atmega328p, atmega2560. For the latter I introduced autoincrement of addresses so that optiboot works beyond the 128kb upload barrier of the STK500V1 protocol if the uploader keeps sending data beyond 128kb (avrdude does). The Chaucer example  now prints 255 kb. Admittedly,  for the "big" chips with all features the compiler needs more than 512 bytes. Still by using virtual boot partition, I can get away with 768 bytes.

To WestfW: feel free to review the attached code and use any or all tricks in there for future releases...

[optiboot.zip](https://github.com/Optiboot/optiboot/files/384440/optiboot.zip)

WestfW, 2018-07-09:
Fixed:
https://github.com/Optiboot/optiboot/commit/31cd92f1c9869f2ee9c383ea4476ef77a02fe8bd

closed 2018-07-09 by WestfW