Optiboot/optiboot #178: UART should be started after blinking LEDs to prevent corruption of serial data

bafbafbaf, 2016-02-07:
Optiboot did not work on my nano 3.0.  After debugging, I found that the first character received from the UART was corrupted by a framing error caused by AVRdude sending data while the nano was coming out of reset and blinking the LEDs (this apparently caused a chain of events in which the watchdog reset the board and the cycle repeated forever).  I moved the UART initialization after the blinking LED function call and all was well.

I also updated verifySpace() to be more robust which also helped get things on track at startup.

void verifySpace() {
  for (;;)
  {
    uint8_t c = getch();
    if (c == CRC_EOP) break;
  }
  putch(STK_INSYNC);
}

WestfW, 2016-02-08:
I don't understand how waiting longer to initialize the uart makes a framing error LESS likely.  Don't you want to initialize the uart as soon as possible, so that even if the host starts sending data while things are blinking you receive them properly?  Or is it an overrun issue where the first command sent overflows the buffers if we're not paying attention?

Why do you think this was a problem on Nano when it hasn't been on Uno/etc - they should be logically very similar?

The avrdude "retry" mechanism is currently a bit useless with arduino, because the RESET manipulation happens outside of the retry loop :-(

bafbafbaf, 2016-02-08:
On 2/7/2016 11:26 PM, Bill Westfield wrote:
> I don't understand how waiting longer to initialize the uart makes a
> framing error LESS likely. Don't you want to initialize the uart as
> soon as possible, so that even if the host starts sending [...]
I would agree with your assessment that this is an overflow, not a 
framing error.  The key here is that you don't want to turn on the uart 
until you are ready to process the data (which you're not while blinking 
the LEDs).

> Why do you think this was a problem on Nano when it hasn't been on 
> Uno/etc - they should be logically very similar?
>
I'm not an Arduino expert so I really can't comment on the different 
boards.  Sorry I'm not much help here.  I did copy a couple lines of 
code from another bootloader which turns on some pullup resistors on the 
RX pin - I don't think this helped though.  The code started working 
once I moved the initialization of the uart.

> The avrdude "retry" mechanism is currently a bit useless with arduino, 
> because the RESET manipulation happens outside of the retry loop :-(
>

The bottom line is I spent several hours messing around with this 
instead of a half hour, and I'm willing to share what I did to make it 
work.  I've attached my copy of optiboot.c for your reference so you can 
see the exact changes I made.  Some changes are cosmetic, so you can 
ignore those.

Best regards,

Brian

#define FUNC_READ 1
#define FUNC_WRITE 1
/**********************************************************/
/* Optiboot bootloader for Arduino                        */
/*                                                        */
/* http://optiboot.googlecode.com                         */
/*                                                        */
/* Arduino-maintained version : See README.TXT            */
/* http://code.google.com/p/arduino/                      */
/*  It is the intent that changes not relevant to the     */
/*  Arduino production envionment get moved from the      */
/*  optiboot project to the arduino project in "lumps."   */
/*                                                        */
/* Heavily optimised bootloader that is faster and        */
/* smaller than the Arduino standard bootloader           */
/*                                                        */
/* Enhancements:                                          */
/*   Fits in 512 bytes, saving 1.5K of code space         */
/*   Higher baud rate speeds up programming               */
/*   Written almost entirely in C                         */
/*   Customisable timeout with accurate timeconstant      */
/*   Optional virtual UART. No hardware UART required.    */
/*   Optional virtual boot partition for devices without. */
/*                                                        */
/* What you lose:                                         */
/*   Implements a skeleton STK500 protocol which is       */
/*     missing several features including EEPROM          */
/*     programming and non-page-aligned writes            */
/*   High baud rate breaks compatibility with standard    */
/*     Arduino flash settings                             */
/*                                                        */
/* Fully supported:                                       */
/*   ATmega168 based devices  (Diecimila etc)             */
/*   ATmega328P based devices (Duemilanove etc)           */
/*                                                        */
/* Beta test (believed working.)                          */
/*   ATmega8 based devices (Arduino legacy)               */
/*   ATmega328 non-picopower devices                      */
[... 908 lines cut ...]
        do {
#ifdef VIRTUAL_BOOT_PARTITION
        // Undo vector patch in bottom page so verify passes
            if (address == rstVect0) ch = rstVect0_sav;
            else if (address == rstVect1) ch = rstVect1_sav;
            else if (address == saveVect0) ch = saveVect0_sav;
            else if (address == saveVect1) ch = saveVect1_sav;
            else ch = pgm_read_byte_near(address);
            address++;
#elif defined(RAMPZ)
            // Since RAMPZ should already be set, we need to use EPLM directly.
            // Also, we can use the autoincrement version of lpm to update "address"
            //      do putch(pgm_read_byte_near(address++));
            //      while (--length);
            // read a Flash and increment the address (may increment RAMPZ)
            __asm__ ("elpm %0,Z+\n" : "=r" (ch), "=z" (address): "1" (address));
#else
            // read a Flash byte and increment the address
            __asm__ ("lpm %0,Z+\n" : "=r" (ch), "=z" (address): "1" (address));
#endif
            putch(ch);
        } while (--length);
        break;
    } // switch
}

RickeyWard, 2016-05-10:
I have the same problem, Nanos will get stuck in a bootloop constantly being reset by the watchdog timer. This modified version of the bootloader doesn't fix it for me though. Problem remains unchanged.

fantafrik, 2017-03-16:
The same issue here.

WestfW, 2017-03-17:
So ... what kind of Nanos are people talking about?  CH340G usb chip, or something else?
Any specific brands/links?

WestfW, 2017-03-22:
(was this a problem that only happened under particular watchdog-related configurations?
Can someone give me an exact configuration that exhibits the problem (fuses, what sketch is loaded, etc?)
(IIRC, the default Nano bootloader (which is NOT optiboot) does not handle the watchdog correctly...)

WestfW, 2018-07-25:
Maybe fixed by https://github.com/Optiboot/optiboot/commit/b109eb2615b570ce326e2664553f5d7c5ae9af26 ?

closed 2018-08-03 by WestfW