Optiboot/optiboot #179: EEPROM code writes at incorrect address.

WestfW, 2016-03-04:
EEPROM read/write address is word-aligned and not byte aligned. I've found this bug on ATmega328P with BIGBOOT enabled. This is due to multiplying the address by two at https://github.com/Optiboot/optiboot/blob/master/optiboot/bootloaders/optiboot/optiboot.c#L565 - is there any way to fix this? (I think some more code is allowed)

jj1bdx, 2016-03-04:
@WestfW An old bootloader code shows in https://github.com/dhylands/projects-old/blob/2ea6bf62f2e3c0176f3121cfa0affa2ad5116cbb/avr/bootloader-stk500/boot.c#L356L358 as follows:
```c
        /* Set address, little endian. EEPROM in bytes, FLASH in words  */
        /* Perhaps extra address bytes may be added in future to support > 128kB FLASH.  */
        /* This might explain why little endian was used here, big endian used everywhere else.  */
```

epccs, 2016-04-12:
Perhaps the EEPROM stuff is in the oven... I'll check back after a while. 

I am trying out Ubuntu 16.04 which has avr-libc (1:1.8.0+Atmel3.5.0-1) and gcc-avr (1:4.9.2+Atmel3.5.0-1).

make atmega328 

... looks good to me ...

But for some reason <avr/eeprom.h> is not getting included with atmega1284, though can see it is in the includes. I'm too much of an armature to figure this out.

make atmega1284

> avr-gcc (GCC) 4.9.2
> ...
> BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Error=2.1%
> avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega1284p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3      -DBIGBOOT    -c -o optiboot.o optiboot.c
> ...
> avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega1284p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3      -DBIGBOOT  -Wl,--section-start=.text=0x1fc00 -Wl,--section-start=.version=0x1fffe -Wl,--relax -nostartfiles -nostdlib -o optiboot_atmega1284p.elf optiboot.o -lc
> optiboot.o: In function `appStart':
> /home/rsutherland/Samba/optiboot/optiboot/bootloaders/optiboot/optiboot.c:856: undefined reference to `eeprom_write_byte'
> /home/rsutherland/Samba/optiboot/optiboot/bootloaders/optiboot/optiboot.c:856: undefined reference to `eeprom_read_byte'
> collect2: error: ld returned 1 exit status
> Makefile:531: recipe for target 'optiboot_atmega1284p.elf' failed
> make: *** [optiboot_atmega1284p.elf] Error 1

stefanrueger, 2016-07-26:
I don't think this is a bug, because the avrdude uploader also uses word addresses for eeprom. Try using optiboot (with BIGBOOT) and download the chip's eeprom contents:

avrdude -q -F -D -p atmega328p -C /etc/avrdude.conf -D -c arduino -b 115200 -P /dev/ttyUSB0 -U eeprom:r:eeprom.1:r

Then copy the eeprom.1 raw file to eeprom.2, modify it somehow, upload it (-U eeprom:w:eeprom.2:r) and then download again and compare. It'll work OK (it did for me!). This only might be considered a bug if the other side thinks byte addresses for eeprom. Do you use an/know of a uploader that sets the addresses as byte addresses for eeprom r/w? If there are some (are there?), clearly then optiboot cannot satisfy both types of uploaders. It has to make a choice. I'd go with the STK 500 V1.0 interpretation of the address as word address. Yes, the STK 500 V1.0 protocol is not great, but at least it's simple and makes very small bootloaders possible.

@epccs: eeprom r/w functions are very much alive and kicking. 4.9.2 creates chip specific libraries for  (chip specific) code, see my comment on https://github.com/Optiboot/optiboot/issues/177

Simply link with -latmega1284p as opposed to -lc (change the Makefile).

Aldod2323, 2018-07-10:
I'm a relative newbie... but these appear to be very timely posts. I'm not sure this is related to my problem. I'm currently modifying optiboot which includes calls to read and write EEPROM (EEPROM_READ_BYTE) and (EEPROM_WRITE_BYTE) and noticed it does not pull the expected data that I've written via a sketch (that very simply writes data at an address in EEPROM (0x03EF) then calls WD enable ("wdt_enable(WDTO_15MS)"  The bootloader returns unexpected results but mostly "FF"s … thereby I'm assuming I'm not reading the correct address.  I do not receive any errors regarding undefined calls for eeprom_read_byte or eeprom_write_byte during the Make.

Aldod2323, 2018-07-10:
Please ignore my comment above, I had a bug in my code that was erroneously overwriting values in EEPROM! :)