Optiboot/optiboot #182: noooob Q / need help

boelle, 2016-05-02:
Hi

I got happy when i came arround this one: http://forum.arduino.cc/index.php?topic=53692.new#new

and of course posted there...  but then thought it might be better to ask here

given that i'm a total noob what are the steps to compile/make a bootloader from within the ide or with the tools that come with the ide?

i need bootloaders for sanguinololu with a 644p running at 20 Mhz and for a 1284p also running at 20 Mhz

hope someone can guide a dummie through it. on ide 1.6.8

vanbwodonk, 2016-05-09:
Maybe this can help you, [](https://github.com/MCUdude/MightyCore)

vanbwodonk, 2016-05-09:
Maybe this can help you, https://github.com/MCUdude/MightyCore

boelle, 2016-05-09:
yep.... thanks a lot :-D

closed 2016-05-09 by boelle