Optiboot/optiboot #184: Does the bootloader "fake" the device signature?

MCUdude, 2016-07-02:
I came across [this Arduino topic](http://forum.arduino.cc/index.php?topic=408442.0), where it seemed like an ATmega328 (non p) got two different device signatures.

This makes my brain hurts; I've created a core where multiple versions of one chip (e.g ATmega644p and ATmega644a) are supported. I've only tested it with an ATmega644p, since that's the one I got. Both the ATmega644a and the 644p uses the same bootloader hex file. Does this mean that the ATmega644a actually return the ATmega644p's device signature when it's requested by AVRdude using a USB to serial converter (I hope not!)?

WestfW, 2016-07-03:
Yes, the optiboot bootloader returns a signature that is hard-coded in the bootloader, rather than the actual signature of the chip.   ie - the bootloader lies.   This was an advantage when the 328 (non-P) first came out.  It is essentially identical to the 328P, but wasn't supported by the compiler used by Arduino at the time, so having it masquerade as a 328P was a good solution.

I'm not sure if that's true of the 644A vs 644p; IIRC there was some version of the 644 that only had a single UART (Ah.  That was 644 vs 644A.  And the both have the same signature, despite the rather serious difference in capabilities.)

MCUdude, 2016-07-03:
ok, maybe you have a solution:

I'm creating a new Arduino core, a rather simple one. It will support the ATmega48, ATmega8, ATmega88, ATmega168 and ATmega328 (including the A, PA, P and V variants). In the boards menu the user select what kind of microcontroller he/she is using (e.g ATmega328) and the variant in another menu option (328p/328pa or 328). 

1. When the user are going to burn the bootloader (using an ISP) the correct version (lets pretend the current chip is an ATmega328 (non p)) have to be selected. If not, avrdude will return a signature error (this is correct, right?)
2. After the bootloader is burnt the ATmega328 (non p) will identify itself as an ATmega328p, so the menu option will now have to be changed to "328p/328pa". If not, avrdude will return a signature error (still correct?)

How can I implement this in the best way possible? I want support for all variants and still be as less confusing for the user as possible.

MCUdude, 2016-07-07:
I got my hands on an ATmega328 (non p version), and inserted it to my Arduino UNO. I wasn't able to burn the regular UNO bootloader because it isn't an ATmega328p, so I had to create a new boards.txt file . After modifying the optiboot makefile (replace atmega328p with atmega328) and compiling I'm able to burn the bootloader to the ATmega328. But the bootloader won't let me continue to use the ATmega328 option I made.

How can I make the bootloader return the microcontrollers actual address?

stefanrueger, 2016-07-26:
Optiboot will use the signature of the chip for which it was *compiled* rather than the signature of the chip it runs on. You can set which chip you compile for (eg, in the Makefile  change the MCU_TARGET) and you can burn that bootloader onto the chip. Voila! Technically, even very similar chips may have slightly different instruction set and/or special register set layout, so it's actually a good idea to always compile for the target chip, and not a slightly different one. But, as WestfW said, the mimicry can be useful in some situations.

stefanrueger, 2016-07-26:
And you can use avrdude's option -F (ignore signature reported by bootloader). Overriding the check per default is not a great idea if you handle different chips all the time, though. The signature check has prevented me from burning the wrong programme onto the wrong chip a couple of times...

MCUdude, 2017-05-26:
Totally forgot this..

closed 2017-05-26 by MCUdude