Optiboot/optiboot #187: Arduino IDE 1.6.11 doesn't link in code for ".init0" section (resetFlags).

gitmeister, 2016-09-05:
Noticed my watchdog code was always failing to provide the reset cause and noticed that "resetFlags" was never getting updated.

Dumped the "map" file with avr-objdump -trhd FIO-WDT.ino.elf > map.txt, and noticed that the ".init0" as well as the function resetFlagsInit() never appear anywhere. This is using the stock "test_reset.ino".

As a test I threw in a call to resetFlagsInit() in the code just to see if it would magically appear in the map file, and it did. Is the optimizer messing up thinking it doesn't have to generate code for ".init0" sections?

gitmeister, 2016-09-06:
Thanks to facchinm from the Arduino project for proposing this fix/workaround: Add `__attribute__ ((used))` to the prototype for resetFlagsInit().

This seems to apply to newer Arduino IDEs (i'm using 1.6.11) that use the new compiler optimizer switch -flto.

ctag, 2016-09-14:
I ran into this problem this week as well. Thank you for bringing the solution back from the other thread.

closed 2016-12-19 by WestfW