Optiboot/optiboot #196: 1284 hex

JeroenIoT, 2017-01-02:
Hi,

Trying to get the atmega1284p up and running (and move my bigger codes to the 1284) and I am wondering if there is any reason why the 1284p hex file is not present as a file in the folders (optiboot_atmega1284p.hex)? 

Probably a simple question for most of you but would like to know before I start to build it myself

Thanks a lot

WestfW, 2017-01-03:
There's a 1284p .hex file in the "release" zip file; I decided that having .hex and .lst files in the "source code" repository was not a good idea, but added some additional .hex files to the release files.
Note that there are several common 1284p board that differ slightly (ie in the choice of the built-in LED pin), and you might be better off using a .hex file from the board definition associated with your specific board, rather than the optiboot "generic" version...

closed 2017-01-03 by WestfW

JeroenIoT, 2017-01-04:
Hi Bill, thanks for your answer. It clarifies the situation for me and will use the default hex file (building 1284 from scratch so I need to do some work anyway). Keep up the good work!