Optiboot/optiboot #198: Load over I2C (instead of UART)

deoqc, 2017-02-22:
I have several Arduinos connected by I2C with a Raspberry Pi. I would like to program them over I2C. 

Is it reasonable to do it (using a modified Optiboot loader)? If so, what would be the path?

---

*ps: This may very well be classified as 'out-of-scope', but I'm searching for enlightenment on this issue.*

njh, 2017-03-06:
I can't see any reference to i2c or TWI in the codebase.
I suspect that this would be a completely different bootloader codebase.

I have a similar interest but with the bootloader as the i2c master - loading the firmware from an external EEPROM.

WestfW, 2017-03-17:
Most of optiboot is dealing with the serial communications protocol.  Replacing that with I2C would essentially be a completely different bootloader.

closed 2017-03-17 by WestfW