Optiboot/optiboot #201: Support for failover

nzfarmer1, 2017-03-21:
I posted this here:

https://lowpowerlab.com/forum/wireless-programming-(aka-ota)/failover/

The ideal is to make a copy of the original program memory and roll it back IF the updated program fails to clear a flag.

However, this requires some smarts at the boot loader level to set the flag; and loop in the background to see if it gets cleared before timeout.

Help needed if this change is warranted.

WestfW, 2017-03-22:
That amounts to tripling the amount of code in the bootloader (save old code to SPI flash, program AVR flash from SPI flash, plus steering logic), so it doesn't really fit withing Optiboot's "keep it small" objective.

nzfarmer1, 2017-03-23:
WestfW, agree, that is an issue.  What if saving the SPI backup and setting the flag was done in program space? The boot loader would then just need to perform a delayed check of the flag; trigger a reboot; and then use the same SPI Read code but with an offset.

Perhaps scope for an alternative larger optional fail over boot loader?  I'd roll it myself, but a heads up on how best to implement the delayed flag checking would be appreciated.

nzfarmer1, 2017-03-28:
Ok, so am guessing by the rush to respond, there's not a huge amount of interest here.

That said, if there are any keen coders out there who can give me a hand writing the code that calls CheckFlashImage with a parameter say X milliseconds after boot, that would be appreciated.

majekw, 2017-03-28:
@nzfarmer1, it's not easy.
If you want to do it only using something inside AVR, options are:
1. Use watchdog. If  application wants to use watchdog it will not work
2. Use some timer and interrupt - same thing, additionally bootloader will be much larger. One possible implementation is very similar to Optiboot's virtual boot partition (replace interrupt vector during programming), another is to move all interrupts to bootloader section to intercept it. If application disable interrupts, it will not work.

So, pure software solution is not possible in general way.
If you control bootloader and application and you can be sure that you don't use watchdog or some timer, it could be doable.

I think, that for all possible programs and malfunctions, it could be possible only using some external watchdog or other similar circuit. Best option would be external watchdog with some status bits accessible for program to check real cpu reset cause and then do something according to this flags. It can be done also sparing some external memory for writing such status byte.

https://www.intersil.com/content/dam/Intersil/documents/x508/x5083.pdf - looks interesting, but lacks of reset cause register (not a real obstacle). I think that you could dig deeper to find better device, but it's a path I would go.

closed 2017-05-09 by WestfW