Optiboot/optiboot #203: Atmega328

msuzuki777, 2017-04-18:
I was able to load Optiboot on Arduino 1.6.12. It seemed to work okay but when I selected Boards: Optiboot on 28pin cpus, then processor: ATmega328, the program apparently looked for a ATmega328p.
ERROR: arvdude: Expected signature for ATmega328P is 1E 95 0F

So apparently looking for wrong signature. I believe ATmega328P and ATmega168s worked okay.
Thanks

WestfW, 2017-04-20:
Are you talking about when you burnt the bootloader, or when you tried to upload to the chip using the new bootloader?
The latter is "historical" and desirable.  Since the 328 and 328p are essentially indistinguishable, and older toolsets did not support the 328, the 328 is burnt with a 328p bootloader, and thereafter identifies as a 328p.  But I'm not sure that the "burn bootloader" will work correctly.

msuzuki777, 2017-04-20:
Yes, I was trying to "burn bootloader" to a 328 chip when I got this error message. I some how got a json file and got BoardManager to load Optiloader v6.2.  
Well, I noticed I have some other options to try out. It looks like you have a WestfW/Optiloader and Optiboot/optiboot on GitHub. Is one of those better than the other?

WestfW, 2017-06-19:
"Optiloader" is a utility that quickly programs an optiboot binary (.hex file) onto assorted Arduinos; it is not a version of optiboot in and of itself...

closed 2017-06-19 by WestfW