Optiboot/optiboot #206: comments in Makefile wrongly state that avrdude needs zeros to be written for unused fuse bits

bkerin, 2017-04-20:
The comment was right before avrdude 6.3 but 6.3 and later correctly require you to write ones, and I believe I've had the verify fail if I don't.  Not sure if optiboot is on an older avrdude or if the verify isn't normally done or what.

WestfW, 2017-04-21:
Optiboot pre-dates Avrdude6.3, so the comments refer to older versions.  avrdude is NOT "included" in optiboot, so it isn't "on" a particular version.  I'm not sure what to do with this sort of backward-incompatible change in "external" tools.

From the Avrdude config files recently, it looks like the fuses are supposed to support "don't care" bit values.  Does that just "not work" yet?

bkerin, 2017-04-22:
On Thu, Apr 20, 2017 at 8:46 PM, Bill Westfield <notifications@github.com>
wrote:

> Optiboot pre-dates Avrdude6.3, so the comments refer to older versions.
> avrdude is NOT "included" in optiboot, so it isn't "on" a particular
> version. I'm not sure what to do with this sort of [...]
I think it works like it should now: IIRC atmel datasheets say you're
supposed to write unused fuses to 1 (unprogrammed).  I think current
avrdude fails on a verify or something you don't do this.  Unfortunately I
think that is the opposite of what older avrdude needed, though I'm already
starting to forget exactly why.  I know I had to change my build system
when I went to 6.3 and the changes seemed good.  But I didn't try to
maintain compatibility with old avrdude versions.

Britton

per1234, 2017-04-24:
Just adding some additional information on this:
Arduino has modified their version of AVRDUDE 6.3 to be backwards compatible with the old fuse values:
https://github.com/arduino/avrdude-build-script/commit/8f26fd16395f2c1a0845b83f5d7bf6037e195c8a
https://github.com/arduino/avrdude-build-script/commit/edfbcc9c110e01fb8985cef42bc83d918dabe351
https://github.com/arduino/avrdude-build-script/issues/2
A patch for this was submitted upstream:
https://savannah.nongnu.org/patch/index.php?9110

WestfW, 2018-07-24:
I changed a couple of the fuse settings in https://github.com/Optiboot/optiboot/commit/a50f47b8acf28ea51e72c08914d8811563dcb869
Pretty much just for my convenience during testing with a "stock" avrdude.
But I still think it's an avrdude problem :-)

WestfW, 2018-09-21:
Need to go through and fix more of the legacy targets, and boards-1.6.txt