Optiboot/optiboot #209: Brick-Code

cimba007, 2017-06-12:
Why does this code brick my Atmega328p (pro mini) with optiboot 16mhz bootloader?

```
// **** INCLUDES *****
#include "LowPower.h"

void setup()
{
    // No setup is required for this library
}

void loop() 
{
    // Enter power down state for 8 s with ADC and BOD module disabled
    LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);  
    
    // Do something here
    // Example: Read sensor, data logging, data transmission.
}
```

AkioFujimoto, 2017-06-12:
Is there a refresh way HVPP?

cimba007, 2017-06-12:
Yeah, just flash the bootloader via ISP .. but for now I found no other solution.

For now I switched to https://github.com/MCUdude/MightyCore

Nice thing it is all precompiled hex and flashable via board-manager and arduino .. the mcudude does seem to have  little delay after reset ad this seems to prevent the problem.

WestfW, 2017-06-14:
So "brick" in this case just means that the bootloader no  longer works?
Does it still flash the pin13 LED three times when you attempt an upload?

cimba007, 2017-06-14:
Yeah, uploading a sketch to a assembled and glued pro mini does mean "brick" for me. I tested the ISP method on a second board.

the pin13 led doesn't flash at all .. my guess is the culprit is the hex-file from the board-manager .. I will double check which file gets uploaded.

jkougoulos, 2018-06-13:
I have a similar issue but on atmega328pb. Could you try using BOD_ON and see if the problem persists? In my case when I try to upload a sketch when the MCU is in power down mode with BOD disabled, MCUSR has BORF = 1 & EXTRF = 1  after the reset, triggering the immediate execution of the application instead of invoking the bootloader in optiboot 6.2 (see also issue #140)

sineverba, 2018-07-06:
I got exactly same beahviour on Arduino Pro Mini 3V3. Don't want open another issues, but If need I will do.
In detail, I'm using MySensors library with his sleep. If ProMini goes in sleep.... unable to re-upload a sketch. I did try also "reset when see %%" method, without any success.

I did try with a FT232 and a CP2102 programmer, by the way.

sineverba, 2018-07-07:
I can confirm that with minicore works. So I think there is some issues with sleep command.

jkougoulos, 2018-07-07:
well, I changed line 486 in optiboot.c to "if (!(ch & _BV(EXTRF)))" instead of "if (ch & (_BV(WDRF) | _BV(BORF) | _BV(PORF)))" to restore the old behaviour and it works now, even when I disable BOD in sleep mode.

sineverba, 2018-07-07:
I'm not able to do a similar edit, or... I can edit a file, but after I have no knowledge ti upload/use it :-)

WestfW, 2018-07-08:
Confirmed.
So apparently, if an external reset occurs while the chip is in a power-down state, the reset cause (MCUSR register) will be reported as "brown out reset" + "external reset", which confuses the bootloader.  I don't know if this is a chip bug, or a feature, or ... something else.
There are a bunch of other MCUSR-related bugs/requests/features in the queue, so I'll add this to the list of things to make sure are working when I attack those.

Here's your code modified to show the additional data...

```
#include "LowPower.h"

uint8_t resetFlags __attribute__ ((section(".noinit")));
void resetFlagsInit(void)  __attribute__((used)) __attribute__ ((naked)) __attribute__ ((section (".init0")));
void resetFlagsInit(void)
{
  // save the reset flags passed from the bootloader
  __asm__ __volatile__ ("mov %0, r2\n" : "=r" (resetFlags) :);
}

void setup()
{
  Serial.begin(115200);
  Serial.print("Begin\nSaved MCUSR was 0x");
  Serial.println(resetFlags, HEX);
  Serial.print("Actual MCUSR is 0x");
  Serial.println(MCUSR, HEX);
  Serial.flush();

  // No setup is required for this library
}

#define SLEEP 1

void loop()
{
  // Enter power down state for 8 s with ADC and BOD module disabled
#ifdef SLEEP
  LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
#else
  delay(8000);
#endif
}
```

majekw, 2018-07-20:
> I can confirm that with minicore works

Because MiniCore uses Optiboot with fix from https://github.com/Optiboot/optiboot/pull/158 already merged.