Optiboot/optiboot #214: __boot_page_*_short functions are hard coded rather than using boot_page_*

joelmuzz, 2017-07-17:
Functions __boot_page_fill_short, __boot_page_erase_short, __boot_page_write_short, __boot_rww_enable_short are in optiboot.c writebuffer rather than the define branching boot_page_* in boot.h which redirects these asm functions to the extended and alternate versions which suit chips with larger memory space and special requirements.

Just trying to port into an unsupported AtMega version and found this. Is there a reason why the boot.h redirections were bypassed? (besides their code being bigger and probably blowing out the 512 bytes)

majekw, 2017-07-17:
This issue is similar to #148.
Personally I wouldn't be redefining boot_page_ macros as in #215 and used something like optiboot_page_  instead. It will make other people messing with code aware that this is not the same thing as in standard headers.

joelmuzz, 2017-07-18:
Yes optiboot_page_ makes it clearer what is going on as you say. I updated my version to that.

WestfW, 2017-07-21:
I like the idea of "defaulting" to the standard boot_page_* functions, but I'm not sure I like the renaming.  The new functions are more "the same thing" than they are "not the same thing" (and it's not like pgmspace.h defines _short() versions.)
Also, the pull request is full of gratuitous white-space changes :-(

joelmuzz, 2017-07-23:
If I make a new version cleaned as you suggest are you interested?

WestfW, 2017-07-25:
It's looking more problematic...

Does your code actually work on a chip where the _extended() forms of the boot_* functions get called?  Currently, the "address" variable is only 16bits, with RAMPZ handled separately when the address is read.  If you were to call one of the _extended() forms that wants a 32bit address and sets RAMPZ, it would overwrite the (already adjusted) RAMPZ with zero from the (implicitly casted) 16bit address...

The existing modification to allow Optiboot to work on 256k memory devices don't work by extending the address to 32bits (which is outside the specs of the stk500 protocol), but by using the STK_UNIVERSAL command to set RAMPZ in a sort-of out-of-band hack.

joelmuzz, 2017-07-28:
Its not the _extended forms, its optiboots _short versions.
SPMCSR @ 0x68 is beyond the range of the OUT instruction on this chip, therefore the original _normal versions which use STS are needed. Thats why avr/boot.h sticks with the slightly less efficient STS instruction I guess.
The redefine branching should be able to do that with SPMCSR address check I think, rather than my current ifdef. Either way it is best if optiboot.c doesnt use its custom _short versions directly.

(from what I can figure out anyway, perhaps there is some other issue at work?)

WestfW, 2018-07-24:
https://github.com/Optiboot/optiboot/commit/808d68513b4046ae665e497294b4c8f7843d3bc7

closed 2018-07-24 by WestfW