Optiboot/optiboot #217: Some more code space reduction

aweatherguy, 2017-09-15:
I have found perhaps another 18 bytes of code size reduction (with atmega328 anyway). This is just a little more optimization of existing code with no change in behavior (except it may run a couple CPU cycles faster). I don't think it will break anything for any supported MCU. The changes are fairly small and should not take  much time to review.

Are you interested? 

If so, how can I communicate the changes? I could do a pull request but don't have permission to create a branch in the repo for that. Or I could post the changes here, or e-mail them to somebody...

NicoHood, 2017-09-15:
@aweatherguy fork the project, commit your changes and create a PR.

aweatherguy, 2017-09-15:
Thanks for the tip -- I'm far from knowledgable on git. PR has been submitted.

closed 2017-09-15 by aweatherguy