Optiboot/optiboot #219: USART double speed mode after leaving bootloader?

chupocro, 2017-10-04:
I had some troubles with UART debugging because my code assumes untouched **UCSR0A** but optiboot leaves **U2X0** set. I was looking if that was documented somewhere in the docs but didn't find any mentions.

I might be a good idea to reset **U2X0** before starting the app - if that is possible considering how bootloader works (starting the code by watchdog reset).

That might not be significant when using Arduino IDE but I am sometimes using optiboot loader with Arduino nano and Pro mini boards + AVR Studio (uploading files with avrdude from command line). That way I sometimes upload 'bare metal' code produced by AVR Studio during the development and when everything is finished then I remove a bootloader and upload the code 'normally' (through ISP).

BTW, I was examining the bootloader source and I can't get how can: `UCSRA = _BV(U2X);`
work when the target is ATmega328 which has UCSR**0**A register and the bit name is U2X**0**?!

I can see the line: `#define UCSR0A	UCSRA` in **pin_defs.h** but it should be the opposite so it could work :-/

chupocro, 2017-10-05:
Sorry, my mistake. I flashed optiboot from an Arduino IDE instead of v6.2 from this repository by accident. This version doesn't change **U2X0**.

I would, however, still like to know how it is possible the code compiles for ATmega328 despite using UCSRA and not UCSR**0**A.

WestfW, 2017-10-05:
Even the stock Arduino Optiboot should be starting the sketch via a watchdog reset that clears U2X0...

> how it is possible the code compiles for ATmega328 despite using UCSRA and not UCSR0A.
In the interest of not cluttering up the main code path with a lot of conditional compilation, It uses "UART_SRA" and similar symbols that are defined in pin_defs.h depending on the compile parameters.  https://github.com/Optiboot/optiboot/blob/master/optiboot/bootloaders/optiboot/pin_defs.h#L34
A few instances of UCSRA/etc still exist where the bits are actually different (initialization of the BRG, for instance.)

closed 2017-10-05 by WestfW

chupocro, 2017-10-06:
My mistake, I thought `#if defined (__AVR_ATmega32__)` is true for ATmega328 as well :-[

I burned the optiboot from Arduino IDE twice and both times there was a garbage received from RS232 until I increased the speed from 19200 to 38400 bd. When I added (in my code) the instruction to clear U2X0 then UART speed was OK. [This](https://www.dropbox.com/s/8f4cn8p965u3xq6/optiboot_atmega328-Mini%20%28Arduino%20IDE%29.zip) is the optiboot bootloader from Arduino IDE which does **not** clear U2X0 (not using watchdog reset?). This is a direct Dropbox link to .zip file, no need to register, just click 'No thanx' if there is pop up. That bootloader is from portable version of Arduino (arduino-1.6.12-windows.zip) located in folder `arduino-1.6.12\portable\packages\arduino\hardware\avr\1.6.20\bootloaders\optiboot`.

Thank you for the reply!

WestfW, 2017-10-07:
That's ... interesting.   the "-mini.hex" version of looks like a "leftover" from very early in the optiboot development.  (ie before I took over.)   There's no such target in the the makefile, and there's not VERSION NUMBER in the binary (which is one of the first things I added!), and it does in fact use the old "app_start" code that just jumps to zero without resetting stuff.  I think if you use the "normal" optiboot_atmega328.hex" file from the same place, you would be OK.

https://github.com/arduino/Arduino/issues/6800 submitted to the Arduino folk.  Although changing something that has "been that way" essentially forever may ... not be a high priority.