Optiboot/optiboot #220: Pin high during flashing.

maxholgasson, 2017-11-22:
Hello,

I connected a esp8266 and an arduino pro mini. When trying to load the sketch to the pro mini it fails as the two controllers are constantly exchanging data over uart.
my idea is now when the bootloader recognizes that a new sketch is going to be uploaded then the bootloder could pull a pin high which then triggers reset on the sp8266 and stops so the uart transmissions.  Is there a way to pull eg pin4 high during flashing a new sketch?

WestfW, 2017-11-23:
This would certainly be possible to do, but it's not any "mainstream" enough for me to think about adding to the main repository.  I think I once modified the LEDBLINK code that happens at reset to leave the LED "on" until the appstart(), which might be a way to do things without causing additional "interference" on IO pins...

maxholgasson, 2017-11-23:
Thank you for your reply. Ok, but anyway good to hear that it would be possible.  Hope I can figure out how as it would be awesome for me

closed 2017-12-06 by WestfW