Optiboot/optiboot #223: convert different options into overridable AVRDUDE_FLAGS

sijskes, 2017-12-11:
pseudocoded:
change:
isp: $(PROGRAM)_$(TARGET).hex
	$(AVRDUDE_ROOT)avrdude $(AVRDUDE_CONF) -c $(ISPTOOL) \
              -p $(MCU_TARGET) -P $(ISPPORT) $(ISPSPEED) \
	$(ISPFUSES) \
	$(ISPFLASH)

into:
AVRDUDE_FLAGS ?= $(AVRDUDE_CONF) -c $(ISPTOOL) -p $(MCU_TARGET)
$(AVRDUDE_ROOT)avrdude $(AVRDUDE_FLAGS) $(ISPFUSES) $(ISPFLASH)

sijskes, 2017-12-13:
and possibly $(AVRDUDE_ROOT)avrdude into

AVRDUDE ?= $(AVRDUDE_ROOT)avrdude
target:
$(AVRDUDE) $(AVRDUDE_FLAGS) $(ISPFUSES) $(ISPFLASH)

WestfW, 2018-02-16:
With the idea that AVRDUDE_ROOT and AVRDUDE_FLAGS could be environment variables?
That seems reasonable...
(gee, github sure made a mess of formatting your post(s); I tried to the first one with some "code tags"; please make sure I did it right.)
I'm not sure I understand what the 2nd comment is supposed to mean; can you clarify?

sijskes, 2018-02-16:
indeed 1st cmnt is for allowing override AVRDUDE_FLAGS from outside makefile.
2nd cmnt is for override complete avrdude executable name from outside makefile.

sijskes, 2018-02-16:
pseudocoded:
change:
```
isp: $(PROGRAM)_$(TARGET).hex
	$(AVRDUDE_ROOT)avrdude $(AVRDUDE_CONF) -c $(ISPTOOL) \
              -p $(MCU_TARGET) -P $(ISPPORT) $(ISPSPEED) \
	$(ISPFUSES) \
	$(ISPFLASH)

```
into:
```
AVRDUDE_FLAGS ?= $(AVRDUDE_CONF) -c $(ISPTOOL) -p $(MCU_TARGET)
AVRDUDE ?= $(AVRDUDE_ROOT)avrdude

...

$(AVRDUDE)  $(AVRDUDE_FLAGS) $(ISPFUSES) $(ISPFLASH)

```