Optiboot/optiboot #226: [Question] Preventing bootloader to exit while receiving data

MCUdude, 2018-02-26:
Hi!
I'm working on an application where I use Optiboot wit ATmega1284p. I've changed the watchdog timeout from [1 second to 4 seconds](https://github.com/MCUdude/optiboot_flash/blob/95c5494589d67cea2d48b049f4deb413832db78e/optiboot_flash.c#L565), and LED_START_FLASHES is sat to 0 in the makefile. I'm also using UART0 for upload with a baud rate of 125000 and an F_CPU of 8 MHz.

I have a MAX232 circuit connected to UART0. I want to be able to upload new firmware to the microcontroller using this interface, but auto reset doesn't work properly with any of the RS232 adapters I've tested (the DTR/RTS lines doesn't behave like they should, and I don't get a falling edge every time I try to upload).

Instead I figured out I may directly jump to the bootloader by running these lines (I know it's kinda ugly), and start transfering data:
```c
MCUSR = 0;
((void (*)())0x1fc00)();
```

With a timeout of 4 seconds I should have plenty of time to get AVRdude running. The problem is that while it receives data over serial, it just exits the bootloader after 4 seconds, and starts its incomplete program. To me it seems like the watchdog doesn't get resetwhile the bootloader receives data.

* How does the watchdog reset routine work while receiving data?
* How can I prevent the bootloader from exiting while reveiving data?
  - Why does this method work with a physical reset (EXTRF), but not when MCUSR = 0?

WestfW, 2018-02-28:
> MCUSR = 0;
> ((void (*)())0x1fc00)();

Be sure to disable interrupts (perhaps in each peripheral?) before jumping to the bootloader like this.  Otherwise, you may have an interrupt occur, and vector off to a page of code that the bootloader upload has just changed or erased.  I think that since Optiboot does not use any interrupts itself, it should be sufficient to do a "cli" instruction or "noInterrupts()" call.

MCUdude, 2018-02-28:
Thanks! `cli()` did the trick!

Turns out that with interrupts enabled the bootloader, receved data was detected with a frame error (FE0). This prevents the watchdog from being reset.

Thanks!

closed 2018-02-28 by MCUdude