Optiboot/optiboot #227: Optiboot with EEPROM Support and Auto-Baudrate

awatterott, 2018-03-01:
Just for information, there is a new Optiboot version with EEPROM support and automatic baudrate detection that fits in 512 bytes. The source code can be found here:
https://github.com/svn2github/transistortester/tree/master/bootloaders

Discussion on German forum [here](https://www.mikrocontroller.net/topic/441133).

WestfW, 2018-04-08:
Noted.  Are you maintaining both the .c version and the .S version in that repository?
It looks like you've done quite a lot of work there.  I'm pretty opposed to switching to asm for the main optiboot code; did you discover anything interesting in the process of your changes?

awatterott, 2018-04-08:
The project is not from me. It is from Karl-Heinz Kübbeler (kubi48) and his mail address is in the documentation.

kubi48, 2018-04-08:
All main features of my optiboot version are supported with the .c and the .S source. You can change to the .c source by adding a  C_SOURCE=1 to the make call.
One main difference is, that the EEprom is supported by default with the .s source. The .c source will support also the EEprom, if the option EEPROM_SUPPORT is set to 1.
Unfortunately you can currently use the Makefile only with a Linux operating system, because some Linux tools are used to automatically determine the start address of the bootloader. Also the fuses are recalculated depending on the actual size of the bootloader. Many AVR types are supported now and most of them are basically tested.   Probably the automatic start address feature can be used also with Windows when installing the cygwin package.

nerdralph, 2018-06-20:
I have autobaud and eeprom read working in 256 bytes:
https://github.com/nerdralph/picoboot/tree/master/arduino

I'll probably switch to a version of my soft UART in order to support 115.2kbps at 8Mhz,

kubi48, 2018-06-21:
Plese note, that the operation with 115.2kbps is not safe to use with the 8 MHz clock, especially if you use the internal clock instead of a crystal.
My optiboot version can generate the code for many AVR-Processors and different clock rates.
The measurement of the serial input can be done with 6 different methods. The method can be changed by setting the BAUD_RATE parameter in the make call below 100.
Here is a example of a make call simular to your solution:

$ make atmega328p F_CPU=8000000 LED_START_FLASHES=0 BAUD_RATE=52
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p  -fno-diagnostics-show-caret -DBAUD_RATE=52 -DLED_START_FLASHES=0   -DSUPPORT_EEPROM=1 -DLED=p -DUART=0 -DSOFT_UART=0 \
        -DUART_RX=p -DUART_TX=p -DF_CPU=8000000 \
        -DHFUSE=hexDE -DLFUSE=hexFF \
        -DBOOT_PAGE_LEN=512 -c -o optiboot.o optiboot.S
--------------------------------------------------------------------------------
Simple Baudrate measurement with time limit implemented in optiboot! (4-bit, CLK/8)
UART Minimum 488 Baud, Error surely less than 4% up to 80.0 kBaud
--------------------------------------------------------------------------------
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p  -fno-diagnostics-show-caret -DBAUD_RATE=52 -DLED_START_FLASHES=0   -DSUPPORT_EEPROM=1 -Wl,--section-start=.version=0x7FFE -Wl,--relax -nostartfiles -nostdlib  -o optibootx.elf optiboot.o -lc
###########################################
Boot Loader start address: 0x7E00 = 32256
###########################################

8000000 Hz operation with Auto-Baudrate configured.
avr-size optiboot.elf
   text	   data	    bss	    dec	    hex	filename
    498	      0	      0	    498	    1f2	optiboot.elf
Requires 1 Boot Pages, 512 Bytes each
BOOTSZ=3
avr-objdump -h -S optiboot.elf > optiboot_atmega328p.lst
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot.elf optiboot_atmega328p.hex

You can see in the comments of this make call, that the autobaud function is only safe to use below 80 kbps. The size of the make call with simplest method (BAUD_RATE=22) is  only 488 Bytes,
the next step for the method (BAUD_RATE=32) will use 492 Bytes.

closed 2021-04-30 by WestfW