Optiboot/optiboot #234: Cannot compile optiboot in Atmel Studio7 for atmega328pb (please help...)

focussing, 2018-07-19:
My situation:
- I have installed Atmel Studio 7 on Windows and build a project which I can debug, build, etc.
- the processor is atmega328pb
- now I need a bootloader
- downloaded the complete "optiboot" directory tree to c:\develop
- double click the xplained328pb.atsln file in C:\develop\optiboot\AtmelStudio
- it successfully builds for one of the targets in the makefile like atmega328 but I need optiboot for atmega328pb, using uart1 and a specific led
- added to makefile the following

```
atmega328pb: TARGET = atmega328pb
atmega328pb: MCU_TARGET = atmega328pb
atmega328pb: CFLAGS += $(COMMON_OPTIONS) $(UART_CMD)
atmega328pb: AVR_FREQ ?= 16000000L
atmega328pb: LDSECTIONS  = -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe
atmega328pb: $(PROGRAM)_atmega328pb.hex
atmega328pb: $(PROGRAM)_atmega328pb.lst
```
- in the build properties added to Build Commandline `UART=1 LED=B2 AVR_FREQ=16000000L atmega328pb` which corresponds to my hardware
- building gives the following errors:
```
------ Build started: Project: bootloader, Configuration: Debug AVR ------
Build started.
Project "bootloader.cproj" (default targets):
Target "PreBuildEvent" skipped, due to false condition; ('$(PreBuildEvent)'!='') was evaluated as (''!='').
Target "CoreBuild" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "C:\develop\optiboot\AtmelStudio\bootloader.cproj" (target "Build" depends on it):
	Using "RunCompilerTask" task from assembly "C:\Program Files (x86)\Atmel\Studio\7.0\Extensions\Application\AvrGCC.dll".
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\develop\optiboot\AtmelStudio\..\bootloaders\optiboot" -f "makefile" UART=1 LED=B2 AVR_FREQ=16000000L atmega328pb 
		make: Entering directory 'C:/develop/optiboot/bootloaders/optiboot'
		avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.1_1750) 5.4.0
		Copyright (C) 2015 Free Software Foundation, Inc.
		This is free software; see the source for copying conditions.  There is NO
		warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
C:\develop\optiboot\AtmelStudio\makefile(521,1): error: recipe for target 'baudcheck' failed
		avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3    -DLED=B2  -DUART=1   -c -o optiboot.o optiboot.c
		process_begin: CreateProcess(NULL, bash baudcheck.tmp.sh, ...) failed.
		make (e=2): Het systeem kan het opgegeven bestand niet vinden.
		make: [baudcheck] Error 2 (ignored)
		In file included from optiboot.c:245:0:
c:\program files (x86)\atmel\studio\7.0\toolchain\avr8\avr8-gnu-toolchain\avr\include\avr\io.h(623,6): warning: #warning "device type not defined" [-Wcpp]
		 #    warning "device type not defined"
		      ^
		In file included from optiboot.c:247:0:
c:\program files (x86)\atmel\studio\7.0\toolchain\avr8\avr8-gnu-toolchain\avr\include\avr\eeprom.h(41,3): warning: #warning "Device does not have EEPROM available." [-Wcpp]
		 # warning "Device does not have EEPROM available."
		   ^
		In file included from optiboot.c:254:0:
C:\develop\optiboot\AtmelStudio\boot.h(126,4): error: #error AVR processor does not provide bootloader support!
		 #  error AVR processor does not provide bootloader support!
		    ^
C:\develop\optiboot\AtmelStudio\boot.h(136,4): error: #error Cannot find SPM Enable bit definition!
		 #  error Cannot find SPM Enable bit definition!
		    ^
		In file included from optiboot.c:264:0:
C:\develop\optiboot\AtmelStudio\pin_defs.h(47,2): error: #error UART == 1, but no UART1 on device
		 #error UART == 1, but no UART1 on device
		  ^
C:\develop\optiboot\AtmelStudio\optiboot.c(306,6): warning: #warning BAUD_RATE error greater than 2% [-Wcpp]
		     #warning BAUD_RATE error greater than 2%
		      ^
		optiboot.c: In function 'main':
C:\develop\optiboot\AtmelStudio\optiboot.c(481,8): error: 'MCUSR' undeclared (first use in this function)
		   ch = MCUSR;
		        ^
C:\develop\optiboot\AtmelStudio\optiboot.c(481,8): info: each undeclared identifier is reported only once for each function it appears in
		In file included from c:\program files (x86)\atmel\studio\7.0\toolchain\avr8\avr8-gnu-toolchain\avr\include\avr\io.h:99:0,
		                 from optiboot.c:245:
C:\develop\optiboot\AtmelStudio\optiboot.c(487,17): error: 'WDRF' undeclared (first use in this function)
[... 223 lines cut ...]
		                         ^
C:\develop\optiboot\AtmelStudio\optiboot.c(922,6): info: in expansion of macro '__boot_page_erase_short'
		      __boot_page_erase_short((uint16_t)(void*)address);
		      ^
C:\develop\optiboot\AtmelStudio\boot.h(206,60): error: 'PGWRT' undeclared (first use in this function)
		 #define __BOOT_PAGE_WRITE         (_BV(__SPM_ENABLE) | _BV(PGWRT))
		                                                            ^
C:\develop\optiboot\AtmelStudio\boot.h(387,25): info: in expansion of macro '__BOOT_PAGE_WRITE'
		           "r" ((uint8_t)__BOOT_PAGE_WRITE),      \
		                         ^
C:\develop\optiboot\AtmelStudio\optiboot.c(939,6): info: in expansion of macro '__boot_page_write_short'
		      __boot_page_write_short((uint16_t)(void*)address);
		      ^
		<builtin>: recipe for target 'optiboot.o' failed
		make: Leaving directory 'C:/develop/optiboot/bootloaders/optiboot'
		make: *** [optiboot.o] Error 1
	Done executing task "RunCompilerTask" -- FAILED.
Done building target "CoreBuild" in project "bootloader.cproj" -- FAILED.
Done building project "bootloader.cproj" -- FAILED.

Build FAILED.
========== Build: 0 succeeded or up-to-date, 1 failed, 0 skipped ==========

```
- to me it lookes that the 328pb device is not recognized... what am I doing wrong?

focussing, 2018-07-20:
It works! Thank you very much. I need the second uart of the 328pb so cannot use the 328p, that's why I really need this.

There is a bug in the solution's Build Events Post-build command line. It should be like this `cp ../bootloaders/optiboot/*.hex .` otherwise the build fails.

Also the following error occurs:
```
C:\develop\optiboot\AtmelStudio\makefile(515,1): error: recipe for target 'baudcheck' failed
		avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3    -DLED=B2  -DUART=1 -B "C:\Program Files (x86)\Atmel\Studio\7.0\Packs\atmel\ATmega_DFP\1.2.209\gcc\dev\atmega328pb" -I "C:\Program Files (x86)\Atmel\Studio\7.0\Packs\atmel\ATmega_DFP\1.2.209\include"   -c -o optiboot.o optiboot.c
		process_begin: CreateProcess(NULL, bash baudcheck.tmp.sh, ...) failed.
		make (e=2): Het systeem kan het opgegeven bestand niet vinden.
		make: [baudcheck] Error 2 (ignored)
```
I don't know how to fix this. Copying the baudcheck.c file to the "studio" directory does not help.

WestfW, 2018-07-27:
Some of this might be fixed by https://github.com/Optiboot/optiboot/commit/70bd967c3cc5e242f09cb4849e2ce396bfc60fe7 (assuming that the 328pb avr-gcc stuff is installed.)
The actual request for official support of 32pb and the 2nd UART I've created a new issue for: https://github.com/Optiboot/optiboot/issues/238

closed 2018-07-27 by WestfW