Optiboot/optiboot #235: avrdude has EEPROM bug

WestfW, 2018-07-24:
Optiboot is about to be tuned enough to include EEPROM support in a 512byte image.
That's swell, but it turns out that AVRDUDE is somewhat inconsistent WRT reading and writing EEPROM - some parts (ATmega328x, for example) use a word-based address in the STK_LOAD_ADDRESS command, and other parts (notably ATmega8) ) used a byte-based address.
This is documented here: https://savannah.nongnu.org/bugs/?48131

There's probably an avrdude.conf based patch, but the short-term issue is that an ATmega8 with optiboot will not interact properly with AVRDUDE.  (It doesn't work with ArduinoISP, either!)

Documented here for the sake of completeness.

MCUdude, 2018-08-07:
So, about getting EEPROM support in a 512B image; how can this be done? I just compared "my" repo to yours in terms of compiled size. Both are compiled with avr-gcc 5.4.0.

`make atmega328 LED=B5 LED_START_FLASHES=2 UART=0 AVR_FREQ=20000000L BAUD_RATE=500000 BIGBOOT=1`

"My optiboot_flash repo":
```
  avr-size atmega328/20000000L/optiboot_flash_atmega328_UART0_500000_20000000L.elf
   text	   data	    bss	    dec	    hex	filename
    560	      0	      0	    560	    230	atmega328/20000000L/optiboot_flash_atmega328_UART0_500000_20000000L.elf
```

Official Optiboot repo:
```
avr-size optiboot_atmega328.elf
   text	   data	    bss	    dec	    hex	filename
    570	    156	      0	    726	    2d6	optiboot_atmega328.elf
```

If i've understood it correctly the *text* field is the actual size of the image in bytes. How can this be reduced to 512 or less?

MCUdude, 2018-08-07:
ON TOPIC:
I'm no avrdude.conf guru, so maybe you know where the EEPROM related bug for ATmega8 (or other devices with byte-based addresses) is located? My Arduino cores includes a separate copy of avrdude.conf, so a patch should be easy to roll out to affected users if we can figure how we can fix the bug.

WestfW, 2022-07-18:
See the conclusion at https://github.com/avrdudes/avrdude/issues/967#issuecomment-1188432577 and related discussion.
TLDR: a bug was introduced in AVRDUDE back in 2006, and since then other code (like optiboot and Arduino as ISP) has compensated, assuming that the buggy code was intentional.)

mcuee, 2022-10-15:
For classic AVRs, this has been fixed in avrdude git main by the following PR.
* https://github.com/avrdudes/avrdude/pull/1046.

It does introduce regression for optiboot_x and optoboot_dx. This will be fixed as well once the requirement is clear.
* https://github.com/avrdudes/avrdude/issues/1120

mcuee, 2022-10-22:
There may still be some issues with four chips (ATmega161 ATmega163 ATmega8515 ATmega8535) and there is a potential fix here.
* https://github.com/avrdudes/avrdude/pull/1141

mcuee, 2022-12-19:
From avrdude side, this issue has been fixed in avrdude git main, which will lead to avrdude 7.1 release soon.

mcuee, 2022-12-20:
From Optiboot side, #357 is a real issue related to EEPROM.
* #357

mcuee, 2022-12-20:
urboot (which is indeed inspired by Optiboot) does achieve this goal within 512B.
https://github.com/stefanrueger/urboot

Tight code: most bootloaders fit into

- 256 bytes albeit without EEPROM read/write support
- 384 bytes with EEPROM read/write support
- 512 bytes with dual-boot, EEPROM read/write support and Chip Erase capability

mirh, 2023-05-14:
Didn't the guy at https://github.com/Optiboot/optiboot/issues/227 (most updated version [here](https://github.com/kubi48/avr-assembler-optiboot/tree/main/optiboot) I think?) already handle EEPROM+SPM?