Optiboot/optiboot #236: Issues with Chaucer sketches

MCUdude, 2018-07-25:
Hi!
I thought I should give one of the Chaucer sketches a try, and loaded the chaucer112k sketch to an ATmega1284P. First of all it doesn't even compile for IDE versions greater than 0022. Seems like these are long forgotten.

It was a quick fix getting it to compile with IDE 1.8.5, but I'm experiencing some issues. Seems to hit me when exceeding 64k flash usage. When the full sketch is loaded it only prints the _knightsTale3_ array, three times! How can we get this to work like intended?

WestfW, 2018-07-25:
I'm not sure that Chaucer112 could have ever worked.  You need pgm_read_byte_far() to get beyond 64k, and probably pgm_get_far_address() as well.

MCUdude, 2018-07-25:
The problem is that I don't know which arrays is stored in the "short" section and which in the "far" section. I did a memory dump after upload, and it turns out that the order (starting from the lowest memory address is knightsTale3, knightsTale2, knightsTale1.

Is there a way to tell the compiler where to store the arrays in progmem?

WestfW, 2018-07-26:
At one time I submitted a set of "64K Problem" issues against Arduino:
https://github.com/arduino/Arduino/issues?q=is%3Aissue+64k+author%3AWestfW
It looks like most of them haven't been specifically addressed.  I'm not sure of their current status; some are likely to have been affected by avr-gcc changes, perhaps?  I'm surprised that there haven't been more "visible" problems; perhaps the people who write programs that exceed 64k just figure out how to deal with it...

The linker scripts control the placement of data items in flash.  For progmem there is:
 `    *(.progmem*)`
I'm pretty sure that means that if you manually provide .section names, they'll go in alphabetic order. (although I would have thought that -fdata-sections would have assigned the knightsTale variables each their own section, and they would have been in a different order than you saw...)

WestfW, 2018-07-26:
Meh.  Those sketches never "worked", beyond being big enough to see if the upload failed.
Aside from not using pgm_read_byte_far(), the code didn't reset the counter in between each section.
https://github.com/Optiboot/optiboot/commit/a2e1f3b69f8d786287055a20e3b1c129bc785704

closed 2018-07-26 by WestfW

WestfW, 2018-07-26:
https://www.avrfreaks.net/forum/ordering-data-progmem
It looks like the linker does NOT order sections alphabetically; it just matches them against the regular expression in the linker script.  You have no more control over the ordering of progmem variables than you do of any other variables.

MCUdude, 2019-06-27:
Sorry to bring this up again, but it would be very helpful if these Chaucer sketches did work.

I'm working on a project where an ATmega1284 is communicating with an audio DSP. The microcontroller also holds the DSP firmware (loaded into the DSP on boot), and the total size exceeds 64k. Preferably I'd like to place the two DSP firmware arrays somewhere high up in flash (close to the bootloader). I know I can use pgmspace_read_far to get the data, but how do I tell the compiler where in progmem to place these two DSP arrays in the first place? I did some research and saw someone referring to __memx. However, I didn't understand much of it, and no real-world examples were provided.

Do you know how I can place an array a known place in progmem so that I can access it with pgmspace_read_far?