Optiboot/optiboot #242: Wrong response to "Read page" command

Elmi77, 2018-08-03:
Not sure if I'm doing something wrong or if it is really a bug: I try to communicate with Optiboot out of an other micro controller, so no avrdude is involved, I'm communicating out of an own application. This is wat I'm doing at the moment and what works well:

`get sync, EOP -> answer InSync, OK`
`get parameter, SW major, EOP -> answer InSync, 6, OK`
`get parameter, SW minor, EOP -> answer InSync, 2, OK`
`enter progmode, EOP -> answer InSync, OK`
`load address, low byte 0x00, high byte 0x00, EOP -> answer InSync and OK`

That's OK so far and conform to the specification from http://ww1.microchip.com/downloads/en/AppNotes/doc2525.pdf

But then it fails with the next command:

`ReadPage, 0x00, 0x80, F, EOP`

According to the specification mentioned above the answer should be

`InSync, 0x00, 0x80, ", <128 bytes of data>, OK`

but I get only

`InSync, OK`

When I use avrdude for communication everything works as expected.

WestfW, 2018-08-03:
> the answer should be
> InSync, 0x00, 0x80, ", <128 bytes of data>, OK

Where do you see the response including the length or quote character?
I only read InSync <page of data> OK

Do you not see any characters after the InSync, or is your code waiting for 132 bytes when there are only going to be 129?   Or something?

Elmi77, 2018-08-03:
OK, finally I found the problem Optiboot responds with

`InSync, <128 bytes of data>, OK`

which seems to be OK.

At the moment I'm struggling with PROG_PAGE, it seems to behave different than described in http://ww1.microchip.com/downloads/en/AppNotes/doc2525.pdf. Here after sending my data

PROG_PAGE, 0x00, 0x7F, 'F', <127 bytes of data>, EOP

as specified there, I never get back a

`InSync, OK`

So may be there Optiboot behaves different than specified?

WestfW, 2018-08-03:
The current optiboot code will go into an infinite loop if you give it an odd number of bytes in the PROG_PAGE command, because the loop termination of the write code is "while (len -= 2)"
Using a length in PROG_PAGE other than the native page size for the chip is probably a bad idea.

Elmi77, 2018-08-08:
I got it working - thanks for your help!

closed 2018-08-08 by Elmi77