Optiboot/optiboot #245: Can't download optiboot 7

deirdreobyrne, 2018-08-16:
You've mixed up your URLs. The json is at https://github.com/Optiboot/optiboot/releases/download/7.0-180815/package_optiboot_optiboot-additional_index.json, the zip is at https://github.com/Optiboot/optiboot/releases/download/7.0-180815/Optiboot-7.0.zip, but the json says the zip is at https://github.com/Optiboot/optiboot/releases/download/v7.0/Optiboot-7.0.zip

Error downloading https://github.com/Optiboot/optiboot/releases/download/v7.0/Optiboot-7.0.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/Optiboot/optiboot/releases/download/v7.0/Optiboot-7.0.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading https://github.com/Optiboot/optiboot/releases/download/v7.0/Optiboot-7.0.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:102)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 404
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:194)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 4 more

Roberto6969, 2018-09-18:
It would be great to correct this issue

[image]

WestfW, 2018-09-18:
Fixed...

closed 2018-09-18 by WestfW