Optiboot/optiboot #247: V7.0 MCUSR is cleared in bootloader

ortegafernando, 2018-08-17:
Hi, I have download last version from github. I haven't compiled anything, I used hex file directly.

I have burnt optiboot bootloader into an arduino pro mini, and after that I have seen led flashing 3 times (so, it is ok).

The MCUSR example doesn't work, always show same information. 

[https://github.com/Optiboot/optiboot/blob/master/optiboot/examples/test_reset/test_reset.ino](url)

I haven't got enough skills to solve this. Anyone could help me? 

Thanks.

ortegafernando, 2018-08-20:
Hi, meanwhile, could any body check the test_reset sketch and see if she/he gets the same result? Thanks

WestfW, 2018-08-21:
What setup are you using what reset methods are you using, and what "same information" are you seeing?
During testing, it was unexpectedly difficult to get certain types of reset, given the way an Uno is set up (ie if you power down the board, you end up powering down the USB/Serial converter as well.)
I mini should be somewhat more controllable, but you still have the problem where the board can get powered (or partially powered) via the logic level on the RxD pin...

ortegafernando, 2018-08-21:
Hi @WestfW  and thank a lot for your answer.

I have do it again. This is the setup.

Arduino Pro Mini (I have got another unit, so I have tried again with this new unit) with new optiboot burned (with an Arduino Uno as ISP)

Upload sketch with and USB/Serial converter and after that disconnect DTR pin and GND pin so Pro Mini is off.

I will put here the serial output after some actions that I have done.

1. Then connect only GND pin:

`Reset flag test

Actual MCUSR content: 0x5 Brownout PowerOn
Passed in R2: 0x5 Brownout PowerOn
Passed in GPIOR0: 0x0

Send 0-9 to set MCUSR, (W)atchdog enable, (J)mp to bootload, (S)leep
`
So this is OK.

2. Press reset button in arduino pro mini

`Reset flag test

Actual MCUSR content: 0x7 Brownout External PowerOn
Passed in R2: 0xF Watchdog Brownout External PowerOn
Passed in GPIOR0: 0x0

Send 0-9 to set MCUSR, (W)atchdog enable, (J)mp to bootload, (S)leep
`
May be I am wrong, but here we have not only "EXTERNAL" but also Brownout and PowerOn.

3. Write W in serial input

`Reset flag test

Actual MCUSR content: 0x7 Brownout External PowerOn
Passed in R2: 0xF Watchdog Brownout External PowerOn
Passed in GPIOR0: 0x0

Send 0-9 to set MCUSR, (W)atchdog enable, (J)mp to bootload, (S)leep
`
Again the same, and nothing about Watchdog.

@WestfW  probably I am doing something wrong, but I have check it with two arduino pro mini's.

May be anybody could try this sketch in another IC's

Or do you want me to do another check? 

Thanks a lot

WestfW, 2018-09-01:
Sorry; I've been out of town more than "near keyboard" for the last couple of weeks...

> here we have not only "EXTERNAL" but also Brownout and PowerOn and Watchdog in R2.

Did you use the "0" command to reset the MCUSR register before you hit reset?  One of the ... features ... of not resetting MCUSR in the bootloader is that its bits are now "sticky" - if you do not explicitly reset them in your sketch, the various reset causes will "add" all the different types of reset that have occurred.
It's assumed that people who want to read MCUSR will understand and deal with this in an appropriate way - the test_reset sketch intentionally only does this manually, so that you can see/check the additive effect.

ortegafernando, 2018-09-01:
Hi @WestfW 

Thanks a lot. I know about MCUSR (a bit), and normally I use it as you says, clearing MCUSR register at the start of my setup.

Normallly I use in my setup (when I have not bootloader):

```
myMCUSR = MCUSR;
MCUSR;
wdt_disable;
```

But with optiboot, I understood that optiboot saves MCUSR in any kind of "r2" register (that i don't know), and optiboot also does all of stuff with MCUSR. 

I understood that we have to read "r2" register as MCUSR is also "contaminated" with the watchdog reset of optiboot.

So I understand it wrong.

Are you saying that we need to use the "0" command to reset the MCUSR register? or the r2 register?

Thanks a lot.

ortegafernando, 2018-09-07:
Hi @WestfW I am fighting with this sketch.

Probably I am not an expert, but I usually work with watchdog and MCUSR.

I have tried this secuence:
(first of all, I have check the flash last two bytes to check that I have optiboot 7.0)
1. power on --> MCUSR and R2 are Ok 
2. write "0" to put MCUSR to 0x00h
3. push reset button -->MCUSR is Ok, R2 includes wathdog
4. write "0" to put MCUSR to 0x00h
5. write "w" to generate a watchdog  --> boot loop until I power off the arduino pro mini.

This is the serial output with that secuence:

```
Reset flag test

Actual MCUSR content: 0x5 Brownout PowerOn
Passed in R2: 0x5 Brownout PowerOn
Passed in GPIOR0: 0x0

Send 0-9 to set MCUSR, (W)atchdog enable, (J)mp to bootload, (S)leep
0

•ÝMCUSR content: 0x0

Send 0-9 to set MCUSR, (W)atchdog enable, (J)mp to bootload, (S)leep

Reset flag test

Actual MCUSR content: 0x2 External
Passed in R2: 0xA Watchdog External
Passed in GPIOR0: 0x0

Send 0-9 to set MCUSR, (W)atchdog enable, (J)mp to bootload, (S)leep
0

•ÝMCUSR content: 0x0

Send 0-9 to set MCUSR, (W)atchdog enable, (J)mp to bootload, (S)leep
w
Reset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüReset flag tesüRe .....
```
What I am doing wrong?

When I work with ATMEGA328P standalone chips, I have this instructions at the begining of the setup:

```
  myMCUSR = MCUSR;
  // Clear all MCUSR registers immediately for 'next use'
  MCUSR = 0;
  wdt_disable();
```

The test_reset.ino sketch hasn't "wdt_disable()" probably because optiboot takes care of it. But then ... why I am getting a boot loop ? I haven't change anything in your sketch, and I have only optiboot 7.0 burnt in my pro mini. 

Regards

ortegafernando, 2018-09-12:
hi, anybody has tried ? thanks

WestfW, 2018-09-14:
>> why I am getting a boot loop ?
I'm getting the boot loop as well.  But not on ATmega8!  Which is particularly mysterious.
I don't see LED flashes within the loop, so I'm not sure it's a "normal" boot loop.
Investigating...
(Sorry for the delay.  For some reason, your earlier updates didn't result in me getting the usual email notification, so I didn't realize that you had updated the issue...)

WestfW, 2018-09-14:
Ahh.  I see.   You can't turn off the watchdog while WDRF is set.
If you set MCUSR to 0, and then jump to the bootloader, it will run the bootloader as intended, but if there is no bootloader traffic, it will do a watchdog reset.  At that point, WDRF is the only bit set in MCUSR, so the startup code bypasses the reset of WDRF (necessary to get support of the application flag in the sketches), but then when it tries to do the WATCHDOG_OFF it fails because WDRF is still set.
Essentially, there's no (easy?) way to tell the difference between "a watchdog timeout occurred during the user sketch" (we want the sketch to restart immediately, with WSRF still set) and "a watchdog occurred in the bootloader after the application requested that the bootloader run" (in which case we want the application to run after disabling the watchdog.)

Unfortunately, that makes perfect sense - in the latter case, the bootloader is essentially being run as a subroutine of the application, so the cases are indistinguishable.

One workaround is to just say that any application that wants to use the bootloader by request must ensure that any application that starts (either the original, or whatever is bootloaded)

WestfW, 2018-09-17:
1) No, I can't think of a way to fix Optiboot itself.
2) your sketch can turn off the watchdog early in setup().    I did update the example to show this.
    It will need to do something else to tell the difference between an actual watchdog timeout, and a watchdog timeout that happened in the bootloader.
3) get *what* working?  The example by itself is pretty meaningless.  The main "problem" is using both watchdog timeout and "bootloader called from the application" in the same application.  "working" would depend on exactly what you want to have happen...

ortegafernando, 2018-09-18:
> 1. No, I can't think of a way to fix Optiboot itself.
> 2. your sketch can turn off the watchdog early in setup().    I did update the example to show this.
>    It will need to do something else to [...]

Ok, I will try and check option 2. Let's see what I get.

Thanks a lot.

WestfW, 2018-09-19:
I've started another "issue" to cover the technical aspects of this bug: https://github.com/Optiboot/optiboot/issues/250
(The discussion can keep going here, though.)

WestfW, 2018-09-22:
Sounds good to me.
The bootloader continues to put mcusr contents in R2 for "backward compatibility", but  you don't need to look at both R2 and the actual MCUSR...
I'll go ahead and close this issue.

closed 2018-09-22 by WestfW

ortegafernando, 2018-09-23:
Thanks a lot. I still don't understand exactly ( ;) ;) R2's stuff but ... ... ... It works for me. (I hope ;)