Optiboot/optiboot #252: OptiBoot not compiling after following instructions on the wiki

macdonaldtomw, 2018-09-21:
I followed the instructions on the Wiki:

https://github.com/Optiboot/optiboot/wiki/CompilingOptiboot

Here is what I get when I copy/paste the example command line invocation listed on that page:

[image]

WestfW, 2018-09-21:
The directory you're using is the original optiboot directory that comes with the Arduino IDE.
Did you actually download the new source code from here?  The code that comes with the IDE is about 7 years and 100+ commits old, and doesn't have the 1284 support.

macdonaldtomw, 2018-09-22:
OK, I'm on my home computer now, also running Windows 10.  Just installed WinAVR fresh.  Fresh clone of Optiboot repo.  Here is my output now:

`macdo@DESKTOP-OK9AC3M MINGW64 /d/PARTICLE LOCAL/optiboot/optiboot/bootloaders/optiboot (master)
$ avr-gcc -v
Using built-in specs.
Target: avr
Configured with: ../gcc-4.3.3/configure --enable-win32-registry=WinAVR-20100110 --with-gmp=/usr/local --with-mpfr=/usr/local --prefix=/c/WinAVR --target=avr --enable-languages=c,c++,objc --with-dwarf2 --enable-doc --disable-shared --disable-libada --disable-libssp --disable-nls --with-pkgversion='WinAVR 20100110' --with-bugurl='URL:http://sourceforge.net/tracker/?atid=520074&group_id=68108&func=browse'
Thread model: single
gcc version 4.3.3 (WinAVR 20100110)

macdo@DESKTOP-OK9AC3M MINGW64 /d/PARTICLE LOCAL/optiboot/optiboot/bootloaders/optiboot (master)
$ make -v
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-pc-mingw32

macdo@DESKTOP-OK9AC3M MINGW64 /d/PARTICLE LOCAL/optiboot/optiboot/bootloaders/optiboot (master)
$ make atmega328
avr-gcc (WinAVR 20100110) 4.3.3
Copyright (C) 2008 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Difference=2.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3            -c -o optiboot.o optiboot.c
optiboot.c:355:6: warning: #warning BAUD_RATE off by greater than 2%
optiboot.c:407: warning: 'leaf' attribute directive ignored
optiboot.c:408: warning: 'leaf' attribute directive ignored
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3          -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o
c:/winavr-20100110/bin/../lib/gcc/avr/4.3.3/../../../../avr/bin/ld.exe: section .version [00007ffe -> 00007fff] overlaps section .text [00007e00 -> 00007fff]
make: *** [optiboot_atmega328.elf] Error 1
rm optiboot.o
`

WestfW, 2018-09-22:
> avr-gcc (WinAVR 20100110) 4.3.3
> Copyright (C) 2008 Free Software Foundation, Inc.
> 
Your compiler is VERY old.  Consider upgrading using the version downloadable from Microchip, or the version packaged with Arduino.

> c:/winavr-20100110/bin/../lib/gcc/avr/4.3.3/../../../../avr/bin/ld.exe:
>    section .version [00007ffe -> 00007fff] overlaps section .text [00007e00 -> 00007fff]
> make: *** [optiboot_atmega328.elf] Error 1

This is the error you get when the compiled code is too big to fit in the 510 bytes that are available.
I guess the do_spm code pushed it over the edge using the old compile, even those the most recent gcc leaves 30bytes  free.

If you want do_spm() and must compile using WinAVR, you could do:
~~~
make atmega328 LED_START_FLASHES=0 LED_START_ON=1
~~~
If you don't need do_spm, you could use:
~~~
make atmega328 NO_APP_SPM=1
~~~

macdonaldtomw, 2018-09-22:
Upgraded to latest version available on Microchip website (5.4.0)

Works now!

```
macdo@DESKTOP-OK9AC3M MINGW64 /d/PARTICLE LOCAL/optiboot/optiboot/bootloaders/optiboot (master)
$ make AVR_FREQ=8000000L BAUD_RATE=38400L LED=D2 atmega328
avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.1_1752) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 38400, Real: 38461, UBRRL = 25, Difference=0.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=8000000L  -DBAUD_RATE=38400L -DLED_START_FLASHES=3    -DLED=D2        -c -o optiboot.o optiboot.c
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328p -DF_CPU=8000000L  -DBAUD_RATE=38400L -DLED_START_FLASHES=3    -DLED=D2      -Wl,--section-start=.text=0x7e00 -Wl,--section-start=.version=0x7ffe -Wl,--relax -nostartfiles -o optiboot_atmega328.elf optiboot.o
avr-size optiboot_atmega328.elf
   text    data     bss     dec     hex filename
    478       0       0     478     1de optiboot_atmega328.elf
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega328.elf optiboot_atmega328.hex
avr-objdump -h -S optiboot_atmega328.elf > optiboot_atmega328.lst
rm optiboot.o optiboot_atmega328.elf
```

Quick question for you:  is this what I should be doing to build optiboot for the ATmega328PB?

Pretty sure everything Optiboot would use (UART, watchdog, EEPROM) is the same with the ATmega328PB as it is for the ATmega328/ATmega328P  right?

WestfW, 2018-09-23:
> is this what I should be doing to build optiboot for the ATmega328PB?

Everything is the same except for the signature that Optiboot "fakes" - your bootloader will report the chip as an ATmega328p.  That was desirable for a while, but now you should probably do:
~~~
make AVR_FREQ=8000000L BAUD_RATE=38400L LED=D2 atmega328pb
~~~

WestfW, 2019-09-16:
You need a recent install of avr-gcc and/or "packs" support for the compiler to recognize the ATmega328pb.

the Makefile should probably support packs in some way.  I'll make that a separate issue.

closed 2019-09-16 by WestfW