Optiboot/optiboot #253: New MCUdude makefile and build script

MCUdude, 2018-09-25:
I was thinking about doing a PR, but quickly relalized I might break something (seems like you've done some small changes compared to my version).

Anyways the new makefile + build script should be considered added to the next optiboot release. Here's whats new comared to the version that Optiboot 8.0 bundles:

* The build script can now automatically create missing folders or delete empty folders from previous builds. The previous script would fail if the folders didn't exist.
* By adding SUPPORT_EEPROM=1 or BIGBOOT=1 eeprom support can be turned on. The output file will have *eeprom_support.hex added to the file name. LDSECTIONS is also automatically changed to match the increased bootloader size.
* All compiled files and corresponding folders are now moved into a new folder: *bootloaders*.
*  *.lst files are now only generated if ASM_OUTPUT=1 is added to the build script
* The build script now generates a [build info text file](https://github.com/MCUdude/optiboot_flash/blob/v7.1/bootloaders/atmega128/atmega128_build_info.txt) for each microcontroller with information for each hex file. The text file is created using **awk**, so it will work on any UNIX compatible environmnet

[You can find the optiboot_flash branch with the updated build script here](https://github.com/MCUdude/optiboot_flash/tree/v7.1).

WestfW, 2018-09-26:
Yes; the main optiboot makefiles didn't retain your "object file" directory  structure, mostly because it's not necessary or convenient for the model of "I want a particular version of a particular target" that the source repository is aimed at.  It's clear that in the case of the MCUDude Cores, the directory structure makes sense, and  I like the fine-naming scheme in general.

I'll see if I can come up with something that will serve both needs.  A change in the directory structure in general is long overdue.

The .lst file avoidance can probably be controlled by the PRODUCTION variable; maybe the subdirectory structure placement can be as well.

BTW, I did some tests with the removal of the baudcheck.sh invocation under PRODUCTION, and it only speeds up the overall build by 10-20%, which is a bit disappointing.  I guess compilation is really expensive, relatively speaking.

MCUdude, 2018-09-26:
> Yes; the main optiboot makefiles didn't retain your "object file" directory structure, mostly because it's not necessary or convenient for the model of "I want a particular version of a particular target" that the source repository is aimed at.

Lets say you want to build four bootloader files:
```
make atmega328p AVR_FREQ=20000000L BAUD=115200 LED=B5 LED_START_FLASHES=2 UART=0
make atmega328p AVR_FREQ=16000000L BAUD=115200 LED=B5 LED_START_FLASHES=2 UART=0
make atmega328p AVR_FREQ=8000000L BAUD=250000 LED=B5 LED_START_FLASHES=2 UART=0
make atmega1284p AVR_FREQ=16000000L BAUD=115200 LED=B0 LED_START_FLASHES=2 UART=0 BIGBOOT=1
```

If these are built using the build script the output structure would look like this:
```
bootloaders/
    atmega328p/
        16000000L/
            optiboot_flash_atmega328p_UART0_115200_8000000L_eeprom_support.hex
        20000000L/
            optiboot_flash_atmega328p_UART0_115200_8000000L_eeprom_support.hex
        8000000L/
            optiboot_flash_atmega328p_UART0_250000_8000000L_eeprom_support.hex
        atmega328p_build_info.txt
    atmega1284p/
        16000000L/
            optiboot_flash_atmega1284p_UART0_115200_16000000L_eeprom_support.hex
        atmega328p_build_info.txt
```

I don't quite see how compiled files located in their respective folders is inconvinient for anyone. Remember that only *atmega328p/* and *atmega1284p/* folder will be created (inside *bootloaders/*).
I'm looking forward to your solution 🙂 Meanwhile I'm going to try adding LED pin and LED start flashes to the build output file, since these options isn't reflected in the file name.

WestfW, 2018-09-29:
> I don't quite see how compiled files located in their respective folders is inconvinient for anyone.

Well, my target workflow is something like:
```
make atmega1284p
avrdude ... optiboot_at<tab for completion>
```
that is - optimized for working with a single target, with defaults to the usual values, and a destination file that is "handy."  Your scheme is great for your bulk builds of 4000 targets, but... doesn't lend it self to easy typing of that second command, and doesn't even work if I want to use some weird 13.5MHz crystal I happen to want to use for some strange reason.

Here's an idea - you currently have directories created by the makeall shell script rather than the Makefile itself - it seems reasonable to either pass the destination directory in as a variable, and/or only use a directory-based target IF that directory exists.
`atmega16: $(DESTDIR)$(PROGRAM)_atmega16_UART$(UART)_$(BAUD_RATE)_$(AVR_FREQ)_eeprom_support.hex`

MCUdude, 2018-09-29:
It would be great if the Makefile itself could create the directories if they don't exist already, and let the makeall script remove all empty folders. I'm no makefile expert, so I'm not quite sure how to implement this in a nice way

MCUdude, 2018-12-07:
I just updated the makeall script and the makefile. Folders are now created by the makefile. This means the makefile can be used even though the bootloaders/atmegaXX folder don't exist.

BTW do you know if it's possible to use the TARGET parameter a build entry? [I'd like to rather refer to $(TARGET) than atmega8](https://github.com/MCUdude/optiboot_flash/blob/c59037f1a643e132fddce9c35ec6129cce2a5b94/Makefile#L218), but if I replace atmega8 with $(TARGET) it just leaves a blank space.

WestfW, 2020-06-03:
...use the TARGET parameter...
Huh.  I'm not sure how those "target specific assignments" actually work.  They're inherited from the original optiboot.  The Make documentation says the assignments are "local", which almost explains it, except for the modified variables that DO seem get set globaly.

It behaves "differently" if you set TARGET on the command line, which might help you in particular...

```
# Makefile.MCUDude modifed:
# atmega88pb: $(PROGRAM)_$(TARGET)_UART$(UART)_$(BAUD_RATE)_$(AVR_FREQ).hex
make atmega88pb LED=B5 TARGET=atmega88pb
  
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex optiboot_atmega88pb_UART__.elf optiboot_atmega88pb_UART__.hex

```

mcuee, 2022-12-22:
The build scripts from optiboot_flash do not work well under Windows as of now, even though they work perfectly fine under Linux and macOS.
* https://github.com/MCUdude/optiboot_flash/issues/8

WestfW, 2022-12-22:
Windows has become somewhat problematic, since the preferred compiler (Atmel/Microchip AVR Toolchain for windows) no longer includes all of the necessary bits (like WINAVR did.)  For example, it no longer includes Make.

mcuee, 2022-12-22:
> Windows has become somewhat problematic, since the preferred compiler (Atmel/Microchip AVR Toolchain for windows) no longer includes all of the necessary bits (like WINAVR did.) For example, it no longer includes Make.

I use git bash plus the Microchip toolchain to overcome this issue.
* https://github.com/Optiboot/optiboot/issues/355#issuecomment-1344967656

mcuee, 2022-12-22:
@WestfW 

If you do not like to introduce git bash in the mix of dependancies, you can try the method from DxCore.
DxCore uses Windows batch file for Windows and shell script for other OS.
https://github.com/SpenceKonde/DxCore/tree/master/megaavr/bootloaders/optiboot_dx

Ref:
https://github.com/SpenceKonde/DxCore/issues/374