Optiboot/optiboot #263: Add atmel ISP feature into optiboot

Harvie, 2019-01-06:
When i need to upload new bootloader to my Atmel, i have to reflash one of my arduinos with ArduinoISP and use that. I've been just thinking about how cool it could be if there was option to compile ArduinoISP directly into the Optiboot, so once in bootloader mode, i could not just upload new .hex to Arduino running optiboot, but also flash .hex using ISP wiring to another atmel. Just instead of saving to local EEPROM it would bitbang the same binary over ISP when this alternative mode is enabled.

I am not sure, how much of impact this would have on final optiboot binary, but it might be handy, if there's way to do this in some minimalistic fashion. This would make any arduino capable of flashing bootloaders without having to upload ArduinoISP sketch, because it would be hidden dormant in bootloader.

WestfW, 2019-03-03:
Like the “optiloader” sketch I have [url]https://github.com/WestfW/OptiLoader[/url], that several people have improved on (lady Ada and Nick Gammon)

closed 2019-03-03 by WestfW

Harvie, 2019-03-03:
That is not what i was talking about.

Optiboot can now store sketch to local EEPROM, but what if there was option to send it using ISP to another arduino? The programmer interface would be the same, only destination would be different.