Optiboot/optiboot #264: Update bootloader from Arduino sketch

Harvie, 2019-01-06:
Lets say i have arduino with bootloader (lets say ATMegaBoot) and i want to flash new version, or completely different binary (eg. Optiboot). Usualy i use ISP programmer to do it. I have to connect programmer to ISP instead of USB and flash the bootloader.

But what if we could just use specialy crafted Arduino sketch to do this? I guess you can accidentaly overwrite bootloader when you are not careful about your sketch code. But what if we made this intentionaly to install new bootloader or install code in bootloader-less mode?

You can upload sketch that would act as ArduinoISP, but instead of bitbanging SPI pins, it would write the bootloader into the beginning of flash memory (given that such sketch would run from the end of flash, it should not be affected. if that is problem, it can copy itself to SRAM or eeprom first).

Harvie, 2019-01-07:
Maybe this is not as easy as i imagined it, but there's partial solution.
The routine to write flash has to be stored in the boot sector. I guess there's already such routine in optiboot. We would only need to slightly modify it, so we can jump to it from sketch and pass sketch data to it.

Problem with this is that it will only work if there's already bootloader, which offers such API.
But for easy bootloader updates it might be perfect, since you will always have old version with this api, which can be used to upload new version with the same api.
if something fails, you will still need to use ISP and users should be warned about this. But that's given by the design from the beginning.

Harvie, 2019-01-08:
Interresting reading:
https://forum.arduino.cc/index.php?topic=392353.0

Harvie, 2019-01-08:
There is even some guy, that achieved just this:

This is the sketch, that calls his modified version of optiboot to reflash the arduino:

https://github.com/majekw/optiboot/blob/supermaster/optiboot/examples/flash_program/flash_program.ino
https://github.com/majekw/optiboot/tree/supermaster
http://forum.arduino.cc/index.php?topic=332191

It would be cool if this could get into optiboot upstream

Harvie, 2019-01-08:
He even claims, that this modification costed only 4 bytes in total optiboot binary size! Very cool!
https://github.com/majekw/optiboot/issues/11

Harvie, 2019-01-08:
And micronucleus bootloader also offers the feature to upgrade itself:
https://github.com/micronucleus/micronucleus/tree/master/upgrade

WestfW, 2021-07-30:
Well, the "write to flash" capability was added some time ago, but it's not clear whether you can actually use the bootloader to overwrite itself, because you'd end up overwriting the vectors that link to the "write to flash" code, and/or the code itself.

Micronucleus self-upgrade happens on chips without a separate boot section, which is easier.