Optiboot/optiboot #265: call to do_sdm in optiboot.h should use EIND on 2560

jandrassy, 2019-01-07:
bootloader on 2560 starts at word address 0x1FE00. the compiler doesn't set EIND, so call to do_spm at atmega 2560, jumps to word address 0xFE01 where FFFF is in empty flash. FFFF is executed as  skip one word if bit set. it is set, so it executes every other FFFF in the rest of the flash until bootloader where it hits the rjmp to do_sdm. it 'works' unless something is written in the flash above word address 0xFE01. 

[https://www.avrfreaks.net/forum/eicall-app-bootloader-atmega-2560-jumps-address-0x1fe01-without-eind-set](https://www.avrfreaks.net/forum/eicall-app-bootloader-atmega-2560-jumps-address-0x1fe01-without-eind-set)

PS: I will not make a PR

WestfW, 2020-05-29:
Ugh.  I guess this is more of a compiler limitation than a bug in optiboot itself.  The code casts a constant integer to a "pointer to function", but pointers to functions are only 16bits on AVRs.  If it were inside of a C program, it would probably be handled via a "trampoline" (a short piece of code in the first 64k that does the full 32bit call/jump, and/or sets EIND, but I guess that doesn't work for the hack of casting the constant...

jandrassy, 2020-05-29:
fix is in https://github.com/Optiboot/optiboot/pull/269

WestfW, 2020-05-30:
Done (copied from https://github.com/Optiboot/optiboot/pull/269 as suggested.)
https://github.com/Optiboot/optiboot/commit/243ee94ffa9f9d9d1949b0664b18e05918a5c03d

closed 2020-05-30 by WestfW