Optiboot/optiboot #275: The jump destination of the patched reset vector is off one word.

ngkz, 2019-08-25:
`RJMP k` means jump to `PC + 1 + k`, not `PC + k`.
You forgot to subtract one from operand when patching reset vector. Because of this, the first instruction of main() won't execute when flash size is <8K and using virtual boot partition.

WestfW, 2019-08-25:
Huh.   You mean main() of optiboot itself, right?

ngkz, 2019-08-26:
>Huh. You mean main() of optiboot itself, right?
Yes

ngkz, 2019-09-08:
I confirmed that the bug fixed. (Tested with ATTiny85)
```
       0:       df ce           rjmp    .-578           ;  0x1dc0
```
```
    1dc0:       01 c0           rjmp    .+2             ;  0x1dc4
    1dc2:       0d c1           rjmp    .+538           ;  0x1fde
    1dc4:       11 24           eor     r1, r1
```

closed 2019-09-08 by ngkz