Optiboot/optiboot #276: How to upload sketch on mega2560

harryberlin, 2019-09-07:
Hello

I did compile `make atmega256` the hex file for atmega2560. 
Moved the File to bootloader path. `../packages/Optiboot/hardware/avr/0.8.0/bootloaders/optiboot/optiboot_atmega2560.hex`
Burning the bootloader in Arduino IDE finished without error.
But after compiling the sketch, it's not uploading the hex-file.

Could you tell me, how to get it working?

WestfW, 2019-09-07:
The normal 2560 bootloader is NOT "optiboot", so you'll need to change various bits in the boards.txt file, or use MCUDude's "Megacore" which already has various things set up for you.

harryberlin, 2019-09-08:
I'd like to give it a try, because with megacore there are missing some bytes by serial communication. so i hope the optiboot can solve this.
which bits i have to change?

WestfW, 2019-09-08:
First, you should probably copy the bootloader to _installDir_/hardware/arduino/avr/bootloaders/optiboot/optiboot_atmega2560.hex - I'm trying to discourage the use of "Optiboot" as a "package", since it doesn't include any core code.  It's an add-on for packages that you should get elsewhere.

Then, try adding the following to _installDir_/hardware/arduino/avr/boards.txt:
```
##############################################################

opti2560.name= Mega 2560 with Optiboot

opti2560.upload.tool=avrdude
opti2560.upload.maximum_data_size=8192

opti2560.bootloader.tool=avrdude
opti2560.bootloader.low_fuses=0xFF
opti2560.bootloader.high_fuses=0xD6
opti2560.bootloader.extended_fuses=0xFD
opti2560.bootloader.unlock_bits=0x3F
opti2560.bootloader.lock_bits=0x0F
opti2560.bootloader.file=optiboot/optiboot_atmega2560.hex

opti2560.build.f_cpu=16000000L
opti2560.build.core=arduino
opti2560.build.variant=mega
opti2560.build.board=AVR_MEGA2560

opti2560.upload.protocol=arduino
opti2560.upload.maximum_size=261632
opti2560.upload.speed=115200

opti2560.build.mcu=atmega2560
opti2560.build.board=AVR_MEGA2560

#####
```

The most important part is probably changing the upload protocol from "wiring" to "arduino"

harryberlin, 2019-09-08:
Thank you. Got it working.

closed 2019-09-08 by harryberlin