Optiboot/optiboot #278: Baud rate error not calculated for mega0/tiny0/tiny1's

MCUdude, 2019-11-10:
This would be a nice feature to get working with the new megaAVRs. I'm wondering what the maximum upload speed when running the CPU at 3.333 MHz, but there's no way of knowing this without manually running the baud rate formulas found in the datasheet.

WestfW, 2019-11-12:
I have myself nearly convinced myself that since the BRG has a fractional divider with a 6bit fraction, the actual rate will always be withing 1/64 (~1.5%) of any desired rate, so I didn't see any value in calculating it.
With a minimum value of 64 (1 and 0 64ths), the max baud rate should be F_CPU/16:  208333bps at 3.333MHz or 166667bps at 2.6667MHz.  Comfortably higher than 115200bps, but not really hitting any of the common rates above that.
(This COULD be a reason for supporting a different clock configuration...)

WestfW, 2020-05-11:
(nearly convinced is now "convinced."  I wrote a quick program to step through all the possible baud rates and calculate the errors using the documented calculations.   Never higher than 1.5%)

closed 2020-05-11 by WestfW